Samsung/TizenRT #92: WIFI_TEST / LWM2M build error

JunhwanPark, 2017-05-15:
Hello,

I set the following to test wakaama in an external function. It is expected to be defined by WIFI_TEST. Do I need to add another defconfig? Or, are there any settings I need to enable for additional?

 * version: 
```bash
$ git hist -2
*   5b52209 2017-05-15 | Merge pull request #90 from Samsung/g_irqvector_sizelimit (HEAD -> master, upstream/master, origin/master, origin/HEAD) [sunghan-chang]
|\  
| * 6e2d1ad 2017-05-12 | Limit the g_irqvector size to NR_IRQS [pradeep.ns]
```
 * branch: master
 * config: artik053/nettest
 * menuconfig: 
   External Functions > DM > LWM2M_WAKAAMA
 * build error:
```bash
CC:  src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c: In function 'prv_dm_conn_scan_result':
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:410:31: error: 'CONFIG_DM_AP_SSID' undeclared (first use in this function)
   WiFiNetworkJoin((uint8_t *) CONFIG_DM_AP_SSID, strlen(CONFIG_DM_AP_SSID), \
                               ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:410:31: note: each undeclared identifier is reported only once for each function it appears in
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:411:58: error: 'CONFIG_DM_AP_SECURITY' undeclared (first use in this function)
       NULL, (slsi_security_config_t *) getSecurityConfig(CONFIG_DM_AP_SECURITY, \
                                                          ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:412:9: error: 'CONFIG_DM_AP_PASS' undeclared (first use in this function)
         CONFIG_DM_AP_PASS, SLSI_WIFI_STATION_IF));
         ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c: In function 'dm_conn_wifi_scan':
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:476:31: error: 'CONFIG_DM_AP_SSID' undeclared (first use in this function)
   WiFiNetworkJoin((uint8_t *) CONFIG_DM_AP_SSID, strlen(CONFIG_DM_AP_SSID), \
                               ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:477:58: error: 'CONFIG_DM_AP_SECURITY' undeclared (first use in this function)
       NULL, (slsi_security_config_t *) getSecurityConfig(CONFIG_DM_AP_SECURITY, \
                                                          ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:478:9: error: 'CONFIG_DM_AP_PASS' undeclared (first use in this function)
         CONFIG_DM_AP_PASS, SLSI_WIFI_STATION_IF));
         ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c: In function 'dm_conn_wifi_connect':
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:510:36: error: 'CONFIG_DM_AP_SSID' undeclared (first use in this function)
   ret = WiFiNetworkJoin((uint8_t *)CONFIG_DM_AP_SSID, strlen(CONFIG_DM_AP_SSID), \
                                    ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:511:61: error: 'CONFIG_DM_AP_SECURITY' undeclared (first use in this function)
          NULL, (slsi_security_config_t *) getSecurityConfig(CONFIG_DM_AP_SECURITY, CONFIG_DM_AP_PASS, SLSI_WIFI_STATION_IF));
                                                             ^
src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:511:84: error: 'CONFIG_DM_AP_PASS' undeclared (first use in this function)
          NULL, (slsi_security_config_t *) getSecurityConfig(CONFIG_DM_AP_SECURITY, CONFIG_DM_AP_PASS, SLSI_WIFI_STATION_IF));
                                                                                    ^
Makefile:55: recipe for target 's5j_dm_connectivity.o' failed
```

sinzah, 2017-05-16:
As I checked build configuration, wifi_test defconfig has 
1023 CONFIG_DM_WIFI=y
1024 CONFIG_DM_AP_SSID="TizenRT1"
1025 CONFIG_DM_AP_PASS="tizenrt_tdc2017"
1026 CONFIG_DM_AP_SECURITY="wpa2_aes"

It should be configurable parameter but, there are no Kconfig including them.
Can we separate Kconfig wakaama from DM framework?

JunhwanPark, 2017-05-16:
@sinzah You are right. 
@dr-venkman I think DM_xxx's define should be in example rather than defconfig. Otherwise, it should be in all defconfig. Did you develop WIFI_TEST? Can you check it?

dr-venkman, 2017-05-16:
@JunhwanPark The error-causing parameters are an integral part of DM (device management) configuration. I could add a #if defined check in source file to see whether these parameters are defined already. If they aren't, an #error statement should provide the necessary guidelines on how to set it up using menuconfig.

JunhwanPark, 2017-05-16:
@dr-venkman Yes, right. I do not see compile errors only when using `artik053/wifi_test`. If I use a different configuration, build errors should not occur.

dr-venkman, 2017-05-16:
@JunhwanPark if I add the #error with the check as mentioned above, you should see the message during build. This will be different from the undeclared variable error you have seen earlier.

dr-venkman, 2017-05-16:
Dear all,

I checked the bug above, reproducing it using the steps sinzah mentioned. 

You have to select and edit the Kconfig  parameters for wifi config. These parameters are found under external/Kconfig file.

Anyhow, I will edit the s5j_dm_connectivity.c file to have the #error in place.

dr-venkman, 2017-05-16:
Dear all,

An update on the issue. We have decided to remove the #error statement, and instead enforce a default value of 'y' to the DM_WIFI config parameter. The config parameters will have default values, so unless you change them, you will most likely see a run-time error.

JunhwanPark, 2017-05-16:
I tested follow version. Build Success! So, I close it. Thanks.

```bash
$ git hist -2
*   5f6afb5 2017-05-16 | Merge pull request #94 from dr-venkman/dm_fixwifi_config (HEAD -> master, origin/master, origin/HEAD) [sunghan-chang]
|\  
| * cbe36ca 2017-05-16 | Automatic enabling of WAKAAMA and DM_WIFI config parameters under DM [dr-venkman]
|/ 
```

closed 2017-05-16 by JunhwanPark