Samsung/TizenRT #118: If I want to add a file into tizenRT filesystem, which directory should I put the file in? Thank you.

cooldo, 2017-05-23:


sunghan-chang, 2017-05-25:
I am sorry for delayed response. Which file or module do you want to add? Due to absence of engineer in charge of file system, we can give an answer this at tomorrow.

cooldo, 2017-05-27:
I am very appreicate your response. :)  thank you very much.
I just want to add a common file to the tizenRT file system. 
for example, I create a file named test.txt in my linux system.  and I want to put this file in tizenRT filesystem.
I don't know which directory I should put this file in,  and then after I compile and burn tizenRT in aritk053, this test.txt can be seen in the file system.
Thank you very much.

Taejun-Kwon, 2017-05-30:
Hi cooldo,
I guess you want use pre-built file, am I right?
Currently TizenRT supports romfs for other boards, but we have to add some codes for ARTIK053.
We will update it soon.
Also we considering fusing tool to add contents inside of smartfs.

cooldo, 2017-05-31:
Yes,  you are right.
When you have the solution for my requirements, would you please share this information to me? Thank you

Taejun-Kwon, 2017-06-02:
Sure,
I will update comment here, Thanks.

kolipka, 2017-06-28:
@cooldo I made simple fix. ROM support is already in TizenRT, but it was not used. 
https://github.com/Samsung/TizenRT/pull/237
You can find simple instruction there how to repartition Artik053 to have small ROM partition. You can also check how auto mounting is done in /mnt case and add similar mechanism.

sunghan-chang, 2017-07-14:
@cooldo Do you have additional question for this? If you don't, let's close this issue.

cooldo, 2017-07-24:
Thank you very much.

closed 2017-07-24 by cooldo