Samsung/TizenRT #142: Can I flash the binary through UART?

qdk0901, 2017-05-31:
Hi,
  Is there anyway to flash the binary with UART? Just like esp8266 or esp32，it can flash binary with openocd through JTAG interface, and it can also flash binary through UART interface. Is there any chance that s5jt200 can flash bianry through UART? Thank you.

sunghan-chang, 2017-05-31:
Hello.

No, you can not flash (program) a binary through UART in sidk_s5jt200 board.
Manufacturer of sidk_s5jt200 is not supporting.

Thank you.

heesub, 2017-05-31:
S5JT200 does not support it. Its ROM code only supports to boot from external flash.

If you want to flash SPI flash through UART, then 'loady', 'loadx', or 'loadb' commands supported by the second level bootloader could be an alternative. Press any key immediately after power-on. Then, U-BOOT shell will appear and you may use them to download binaries on the flash. Please refer to the partition layout defined at /build/configs/artik053/tools/openocd/partition_map.cfg for detailed flash offset where to download an image. For example, "loady 0x040C8000" will download and flash a binary on the OS partition.

sunghan-chang, 2017-06-08:
If you don't have any question, let's close it.

closed 2017-06-08 by qdk0901