Samsung/TizenRT #143: How to build IoTivity in '/external/iotivity'?

DeseDad, 2017-05-31:
I found there are IoTivity code in '/external/iotivity' but it is not built even if I enabled it.
How can I build iotivity?

lssgood, 2017-05-31:
That's an old version of IoTivity. We'll upload the latest verion by this Friday.
And We'll provide a basic guideline to build it.

DeseDad, 2017-05-31:
Thanks  a lot

JunhwanPark, 2017-06-02:
I also wanted to know this. I am expecting an update today.

DeseDad, 2017-06-07:
Is there any update?

lssgood, 2017-06-07:
@DeseDad You can check the PR-#155. Now, we're on the internal review process. After we're done with the review, we'll merge it.

lssgood, 2017-06-12:
@DeseDad We merged it. If you don't have any more questions, please close this issue. Thanks.

DeseDad, 2017-06-13:
Thanks a lot

closed 2017-06-13 by DeseDad

lssgood, 2017-06-13:
@DeseDad If you use iotivity build config(os/tools/.configure.sh artik053/iotivity), you can see 'simpleServer' TASH COMMAND to run the basic iotivity server.
Since we didn't include the tinycbor lib in the extlibs folder, it causes build error. We'll fix it.