Samsung/TizenRT #159: How to understand the pre-built configs for artik053?

laralijuan, 2017-06-06:
There are several provided configs under build/configs/artik053 with different pre-built configurations.
I am confused that what are their purpose? Why not provide one default config with support the artik053's all the HW features for end user?

Thanks very much ahead for your help~

JunhwanPark, 2017-06-07:
Hi @laralijuan 
* tash : for HW feature test
* nettest : tash + network feature test
* extera/typical/minimal : not use for customer. We are not recommand use it.

In my opinion, developers who originally purchased the ARTIK053 board should use nettest. Basic network functions and HW functions are enabled.
You can make your own preferences using menuconfig. For more information on menuconfig see:
https://github.com/Samsung/TizenRT#configuration-sets

laralijuan, 2017-06-08:
Thanks your explanations @JunhwanPark 
And a further question: As artik053 supports fixed features，is it possible to provide a “biggest” config which contains all the features that the artik053 supports. So for the user doesn't need to identify the difference between the multi configs. Is this idea resonable?

JunhwanPark, 2017-06-08:
I appreciate your interest. @laralijuan 

We thought there might be people who do not need networking. So we distinguished between 'nettest' and 'tash'. Like your opinion, developers may feel uncomfortable about this. But we wanted to use them differently if they were the developers who bought this board.

The policy for config will talk a bit more inside the team. Thanks for the good feedback.

laralijuan, 2017-06-13:
If any further info, please share them here. Thanks ~

closed 2017-06-13 by laralijuan