Samsung/TizenRT #186: iotivity some questions

JunhwanPark, 2017-06-13:
I have compiled iotivity.

The version I tested was:
```bash
~/TizenRT/os$ git hist -3
*   a9b9211 2017-06-13 | Merge pull request #183 from jeongchanKim/kernel_gitignore (HEAD -> master, origin/master, origin/HEAD) [sunghan-chang]
|\  
| * f9b4f96 2017-06-13 | Add the file lists in .gitignore  which should be ignored for about built-in apps [jc_.kim]
|/  
*   3c94015 2017-06-13 | Merge pull request #180 from heejin-kim/nw_iotivity [lssgood]
```

I used config as `artik053/iotivity`.

1) Every time I make iotivity, I see that I always create new object files. Is this correct?
2) After make clean, I saw some object files. After clean, I think all object files should be deleted.
```bash
~/TizenRT/os$ find .. -name *.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_subscribe.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_coap_list.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_encode.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_debug.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_option.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_resource.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_str.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_block.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_async.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_hashkey.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_pdu.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_net.o
../external/iotivity/iotivity_1.2-rel/resource/csdk/connectivity/lib/libcoap-4.1.1/libcoap_uri.o
```

Thanks for porting iotivity.

sunghan-chang, 2017-06-13:
@lssgood @heejin-kim Please check this issue.
@JunhwanPark Thank you for sharing issue.

heejin-kim, 2017-06-14:
I think in process of changing examples location(external/iotivity/apps to apps/examples), some issues caused. I'm going over thoes code, then I will commit soon.

JunhwanPark, 2017-06-20:
3. There is an unnecessary content after the build. Delete or add gitignore.
```bash
os$ git st
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../external/iotivity/iotivity_1.2-rel/extlibs/mbedtls/mbedtls/

nothing added to commit but untracked files present (use "git add" to track)
```

sunghan-chang, 2017-07-17:
@JunhwanPark We merged a change to remove un-removed objects through outside Makefile of iotivity. It is not an appropriate method. But we have a plan to update Iotivity so that we will check this with that.

JunhwanPark, 2017-07-17:
@sunghan-chang  Yes! I know. I was always monitoring. :) I think maybe everyone was worried and got the results. Everybody did their best.
Perhaps other questions I would like to know will be solved in a new version. Right? I'm very excited about Iotivity 1.3. Thanks.

heejin-kim, 2017-08-02:
1.Every time I make iotivity, I see that I always create new object files. Is this correct?
- Correct.
2.After make clean, I saw some object files. After clean, I think all object files should be deleted
- Solved.
3.There is an unnecessary content after the build. Delete or add gitignore.
- Solved.
4.When building iotivity, I see a lot of 'warning'. Why do not you remove all the errors by adding the  -Werror  option?
- Checking the warning status and to solve them, it is necessary. And most of warning has been settled.

JunhwanPark, 2017-08-02:
Hi @heejin-kim 
I just updated the source. This is the version I used.
```bash
$ git hist -4
*   0bf580c 2017-08-02 | Merge pull request #336 from thapav/fixAPIerrors (HEAD -> master, origin/master, origin/HEAD) [sunghan-chang]
|\  
| * 6f91a66 2017-08-02 | libc/math: Fix implementation of tanh(),tanhf(),tanhl() API [Vidisha Thapa]
|/  
*   ae5e4e4 2017-08-02 | Merge pull request #313 from lokeshbv/code_coverage [sunghan-chang]
|\  
| * 057d4d0 2017-07-21 | TC: Expand tc for libc/semaphore modules [Lokesh B V]
```

1. What I was wondering was why the object file was already created and why it was re-created. Other files in TizenRT are not. Only iotivity files are always creating new objects. This greatly increases build time. Maybe it is a policy issue of iotivity.

2. OK. I tested.

3. I use 'artik053/iotivity'. After compiling, I checked git status.
```bash
os$ git st
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	../external/iotivity/iotivity_1.2-rel/extlibs/mbedtls/mbedtls/

nothing added to commit but untracked files present (use "git add" to track)
```
I think you should add the following.
```bash
os$ git diff
diff --git a/external/.gitignore b/external/.gitignore
index 4b32ec6..8bc0dd7 100644
--- a/external/.gitignore
+++ b/external/.gitignore
@@ -8,3 +8,4 @@
 /*.adb
 /*.lib
 /*.src
+/iotivity/iotivity_1.2-rel/extlibs/mbedtls/mbedtls/
\ No newline at end of file
```

4. I know that tizenRT's default build policy is 'no warning'. So I think ported iotivity to tizenRT should also have no 'warning'. Of course not necessarily.

I'm very excited about iotivity 1.3. I want to know when the 1.3 version is uploaded. I hope there will be many patches for iotivity in version 1.3.

JunhwanPark, 2017-08-03:
Most of the questions about iotivity were solved. So I close this issue.

closed 2017-08-03 by JunhwanPark