Samsung/TizenRT #195: fail-fast rule violation and developer unfriendly environment

kolipka, 2017-06-14:
TizenRT is not convenient for developers. There is common pattern that in case of missing configuration flag fakes API:
https://github.com/Samsung/TizenRT/blob/master/framework/src/iotbus/iotbus_spi.c#L171

As a result, newcommer instead of getting error during compile time, have to compile, flash, execute code to get information that something is wrong. Usually it will be NULL from open function and that holds no clue what actually happend(wrong config or not supported). This is quite frustrating. 

To detect that API is not supported you have to (read TizenRT sources first) call function other than open with NULL as handle and check whether response is IOTBUS_ERROR_NOT_SUPPORTED or IOTBUS_ERROR_INVALID_PARAMETER. I can imagine much better ways to discover that...

Solution:
remove fake API calls. This will cause developer code to fail during compilation, and enforce proper guarding code behind the configuration flags in consistant way.

JeongJunSik, 2017-06-15:
I think your opinion is right.
However, developers may want to reuse code on multiple platforms without compilation errors.
We need to discuss this internally.

sunghan-chang, 2018-03-29:
@kolipka I already modified what you say in iotbus, https://github.com/Samsung/TizenRT/commit/4eac5d43b24a787a1a485101022ba930da5433a3 and other modules as well. But this your opinion is very good to us so that I want to keep open and use this as reference like above.
Thank you very much.

kolipka, 2018-03-29:
You are welcome :)