Samsung/TizenRT #196: Tools/examples not available despite being configured

tadziopazur, 2017-06-14:
Configuring an example/tool/system command results in the relevant function being compiled int the TizenRT binary, nothing more.

For some functions it might be enough, for most it isn't. All functions of the proto: 
int function(int argc, const char **argv)
make little sense if they are not available through the built-in shell (nsh/tash).
There should be a way (other than adding some code to register each of these functions to tash manually) to have them available after being configured.

sunghan-chang, 2017-06-16:
Hello @tadziopazur .
Does "configuring an example/tool/system" mean executing ./configure.sh?
I am sorry but I don't know what you want exactly.

1. If you want to enable and execute example or tool or system.
enable them through menuconfig and set them as entry (CONFIG_USER_ENTRYPOINT)

2. If you want to add them as TASH command,
use [builtin function](https://github.com/Samsung/TizenRT/blob/master/apps/builtin/README.md).
Almost examples were added, systems will be by us.

Thanks
Sunghan

sunghan-chang, 2017-06-16:
The list of configuration by executing configure.sh is getting default configurations set. That is just a one example (This does not mean apps/examples). We can not provide all of set for examples, systems and tools. Get one default configurations set and modify it as you want.

closed 2017-10-31 by sunghan-chang