Samsung/TizenRT #197: up_i2cuninitialize symbol missing

tadziopazur, 2017-06-14:
The up_i2cuninitialize function is required by i2c tool, that is in the source tree, i.e.:
https://github.com/Samsung/TizenRT/blob/master/apps/system/i2c/i2c_bus.c#L105
and other files in the same directory.

pmarcinkiew, 2017-06-14:
Could you look at this PR?

https://github.com/Samsung/TizenRT/pull/134

sunghan-chang, 2017-06-16:
Hello @tadziopazur .

Did you meet compilation error?
Could you let us know which configuration you use?

tadziopazur, 2017-06-20:
[Configuration file](https://github.com/tadziopazur/TizenRT/blob/i2ctool_build_failure/build/configs/artik053/dot_i2ctool_defconfig)

Set up with:
```
$ cd os/tools && ./configure.sh artik053/tash
$ cp ../../build/configs/artik053/dot_i2ctool_defconfig ../.config
```

Build
```
$ cd ..; make
```

The last lines of the output:

> /opt/iotjs/tizenRT/tizenrt_github_check/os/../build/output/libraries/libapps.a(i2c_bus.o):
>  In function `i2ccmd_bus':
> /opt/iotjs/tizenRT/tizenrt_github_check/apps/system/i2c/i2c_bus.c:105: undefined reference to `up_i2cuninitialize'
> /opt/iotjs/tizenRT/tizenrt_github_check/os/../build/output/libraries/libapps.a(i2c_dev.o): In function `i2ccmd_dev':
> /opt/iotjs/tizenRT/tizenrt_github_check/apps/system/i2c/i2c_dev.c:227: undefined reference to `up_i2cuninitialize'
> /opt/iotjs/tizenRT/tizenrt_github_check/os/../build/output/libraries/libapps.a(i2c_get.o): In function `i2ccmd_get':
> /opt/iotjs/tizenRT/tizenrt_github_check/apps/system/i2c/i2c_get.c:194: undefined reference to `up_i2cuninitialize'
> /opt/iotjs/tizenRT/tizenrt_github_check/os/../build/output/libraries/libapps.a(i2c_set.o): In function `i2ccmd_set':
> /opt/iotjs/tizenRT/tizenrt_github_check/apps/system/i2c/i2c_set.c:214: undefined reference to `up_i2cuninitialize'
> /opt/iotjs/tizenRT/tizenrt_github_check/os/../build/output/libraries/libapps.a(i2c_verf.o): In function `i2ccmd_verf':
> /opt/iotjs/tizenRT/tizenrt_github_check/apps/system/i2c/i2c_verf.c:238: undefined reference to `up_i2cuninitialize'
> Makefile:187: recipe for target '../build/output/bin/tinyara' failed

tadziopazur, 2017-06-20:
I have a workaround the problem in our application code:
```
#if (defined CONFIG_I2C)
#if (defined CONFIG_ARCH_BOARD_ARTIK053) || (defined CONFIG_ARCH_BOARD_SIDK_S5JT200)
// Forward declaration.
struct i2c_dev_s;

int up_i2cuninitialize(FAR struct i2c_dev_s *dev) {
    return s5j_i2cbus_uninitialize(dev);
}
#endif // artik05x
#endif // defined CONFIG_I2C
```
but that's just a workaround..

girishks2000, 2017-06-20:
@tadziopazur
  If its just a work around no issue. But please don't merge this work around. This will break protect build configuration

tadziopazur, 2017-06-21:
Around no issue. What should I make out of it -- are you trying to say it has been fixed by now, or that it is not an issue at all that there is an app in the tree that cannot be built?
Or something else?

sunghan-chang, 2017-09-20:
@tadziopazur You already modified it. Right?  Is there any another issue related this?

closed 2017-10-31 by sunghan-chang