Samsung/TizenRT #202: SPI LSB/MSB API feature broken

kolipka, 2017-06-16:
SPI allows sending data with LSB / MSB feature(_iotbus_spi_s->lsb). This is set here:
https://github.com/Samsung/TizenRT/blob/master/framework/src/iotbus/iotbus_spi.c#L94
but not handled properly:
https://github.com/Samsung/TizenRT/blob/master/os/arch/arm/src/s5j/s5j_spi.c#L351.
As a result board crash. This is a bug.

JeongJunSik, 2017-06-27:
The person in charge is checking this issue.

JunhwanPark, 2017-07-17:
@JeongJunSik In the SPI of artik053 and sidk_s5jt200, there is no part where LSB and MSB can be set. Iotbus_spi needs to be modified.

JeongJunSik, 2017-07-17:
OK. we need to remove it of iotbus_spi API.
I'll upload it after i make changes.

JeongJunSik, 2017-07-17:
It was reflected in #286 
If you do not have any more questions, please close this issue.

kolipka, 2017-07-17:
Well, in my opinion this is quite basic feature of SPI interface and should be implemented. It is relatively easy to make at the software level - you just [need to change bit order](https://stackoverflow.com/questions/746171/best-algorithm-for-bit-reversal-from-msb-lsb-to-lsb-msb-in-c). I think it would be better to implement it in TizenRT than implement workarounds in each external application for TizenRT.