Samsung/TizenRT #212: ISP driver errors

iangcarroll, 2017-06-25:
Hi,

I'm trying to use the ISP driver in TizenRT on the Artik 053 but am hitting some errors. For example, `isp_write_storage` returns `ERROR_SSTORAGE_SFS_FWRITE`:
```c
int a = isp_write_storage("hello", strlen("hello"), 0);
printf("Result ==> %08x\n", a);
```

`ERROR_SSTORAGE_SFS_FWRITE` is also returned with `isp_aes_generate_key_securekey`:
```c
a = isp_aes_generate_key_securekey(16, 0);
printf("Result ==> %08x\n", a);
```

There's no documentation so I don't know if I'm supposed to initialize anything.

joohwani-kim, 2017-06-26:
Current, we are not supporting secure storage. It is necessary to mount the file system for secure storage.
And, I don't recommend access to isp driver directly. 
We are supporting through os/net/tls/see_api.c file.

sunghan-chang, 2017-07-14:
@iangcarroll Do you have additional question for this? If you don't, let's close this issue.

iangcarroll, 2017-07-15:
Is it possible to document how to mount the secure storage file system? The Artik module prominently mentions the security features built into the chip and it'd be helpful to be able to use them.

joohwani-kim, 2017-07-17:
Sorry, we can not provide it in advance.
We will have a release plan for secure storage after internal check.
I recommend to use it after next version release.

sunghan-chang, 2018-02-12:
We supported this functionality serveral months ago. You can find CONFIG_ARTIK05X_AUTOMOUNT_SSSRW on board selection of menuconfig. I close this issue.
If you have any issue, please let us know at anytime.

closed 2018-02-12 by sunghan-chang