Samsung/TizenRT #271: Iotivity *simpleServer* hangs up artik board

awlodarczyk-s, 2017-07-12:
Hello!
I encountered an issue with iotivity on TizenRT. 
Exemplary iotivity server *simpleServer* hangs up artik board.
I started *simpleServer* on artik053. After this, there was no ping response from artik.
Version used: 54cabf8f57166db89d760855b99497ca202ea71a
Configuration: artik053/iotivity

heejin-kim, 2017-07-14:
Hello.
Such a phenomenon has never happened. Also without log, I cannot advise..

Check out those things.
1. Follow test steps. Refer to https://github.com/Samsung/TizenRT/tree/master/apps/examples/iotivity_simpleserver
2. Did you connect Wi-Fi working well?(Server side/ artik053)
3. What command did you typed in Client(Linux) side? ex. ./occlient -t 1

If the same thing happens again, please send me the log for specific debugging.

awlodarczyk-s, 2017-07-14:
Hi.
Actually thinks unexpectedly changed (without any known change in my setup???). Currently *simpleServer* starts successfully then goes through *Discovery* process successfully, but can’t withstand  *“Nonconfirmable Get Request”* from client. The second causes crash with core dump on artik053. The core dump shows that stack of *iotivity_simpleserver* and stack of *IoT_Retransmit* reached its limits.

Commit: 54cabf8f57166db89d760855b99497ca202ea71a

Logs from client and server:
[crash-artik053_simpleclient-discovery.txt](https://github.com/Samsung/TizenRT/files/1148507/crash-artik053_simpleclient-discovery.txt)
[crash-artik053_simpleServerupondiscovery.txt](https://github.com/Samsung/TizenRT/files/1148508/crash-artik053_simpleServerupondiscovery.txt)

heejin-kim, 2017-07-19:
If it caused due to the stack size, you can change those things in menuconfig.

1. IoT_Retransmit thread
~/TizenRT/os$ make menuconfig
>External Functions > IOTIVITY Config Parameters
Default pthread stack size for the **retransmission pthread**

2. iotivity_simpleserver thread
We did not support a config patameter for this stack. So please change the code.
~/TizenRT/apps/examples/iotivitysimpleserver/simpleServer.c
# define **IOTIVITY_TEST_STACKSIZE** <size>

sunghan-chang, 2017-07-31:
@awlodarczyk-s is this solved by modifying stack size? If you don't have additional question, let's close this.

awlodarczyk-s, 2017-07-31:
Unfortunately modification of stack size does not fix the issue.

heejin-kim, 2017-08-02:
We cannot help you unless you give us more information. Attach the log file. 
Also we did not reproduce the situation. I think it is better to debug using GDB in your PC.

sunghan-chang, 2017-08-10:
@awlodarczyk-s Could you share your status?

closed 2017-08-28 by sunghan-chang