Samsung/TizenRT #272: Romfs is not yet supported

daeyeon, 2017-07-13:
Romfs seems not supported yet. Can anyone let me know the status?

sunghan-chang, 2017-07-14:
ARTIK053 supports, SIDK is under internal reviewing.
Please refer #118 for ARTIK053.

daeyeon, 2017-07-14:
Although ROMFS itself is supported, I still feel inconvenient to use it. I need to manually set up the steps 4,5, and 6 as described in https://github.com/Samsung/TizenRT/pull/237. That's a troublesome that affects work speed. Do you have any plan or progress to resolve this difficulty?

sunghan-chang, 2017-07-14:
No, we don't have. That sequence is designed by manufactorer. It is difficult to modify by us.

sunghan-chang, 2017-07-31:
@daeyeon Please check README on doc folder for ROMFS. For SIDK, you don't need to do step 4,5 and 6 like ARTIK053. But for ARTIK053, those are needed. And we don't have a plan to modify them. If you don't have additional query, let's close this.

closed 2017-08-07 by sunghan-chang