Samsung/TizenRT #273: shutdown provided twice

tadziopazur, 2017-07-13:
Both 
os/net/socket/bsd_socket_api.c
lib/libc/net/lib_shutdown.c
provide "int shutdown(int, int)"
under CONFIG_NET flag.

lib/libc/net/lib_shutdown.c has the function gated by CONFIG_SUTOM_SOCKETS flag.
The implementation is empty.

What is this for?
What is the flag - I could not find it anywhere in the TizenRT sources?

The end result is for me that linking fails due to duplicate symbol.

pillip8282, 2017-07-14:
The config parameter you mention should be CONFIG_CUSTOM_SOCKETS,. The apparent typo explains why we have a duplicate symbol in the first place.
Regardless, I think  we do not plan to support shutwodn in libc/net/lib_shutdown.c, as we handle the functionality from bsd_socket_api.c.

tadziopazur, 2017-07-14:
Posted a PR #276

sunghan-chang, 2017-07-14:
Your changes are merged. Let's close this issue. Thank you.

closed 2017-07-17 by tadziopazur