Samsung/TizenRT #279: pthread_cond_broadcast + pthread_cond_wait BROKEN

lucaing80, 2017-07-16:
PThread conditional variables do not work: I was unable to create a working producer / consumerS paradigm without a deadlock. The same code works perfectly on Linux.

Example:

// Global variables:
pthread_mutex_t mut;
pthread_cond_t cond;

pthread_t producer_th, consumer1_th, consumer2_th;
char sndbuf[30];

// Thread functions:
void *consumer1_threadfunc(void *par)
{
	char locbuf[30];

	printf("CONSUMER1 START\n");
	while (true)
	{
		pthread_mutex_lock(&mut);
		{
			pthread_cond_wait(&cond, &mut);
			memcpy(locbuf, sndbuf, sizeof(sndbuf));
		}
		pthread_mutex_unlock(&mut);
		printf("CONSUMER1: %s\n", locbuf);
	}
	
	return NULL;
}

void *consumer2_threadfunc(void *par)
{
	char locbuf[30];

	printf("CONSUMER2 START\n");
	while (true)
	{
		pthread_mutex_lock(&mut);
		{
			pthread_cond_wait(&cond, &mut);
			memcpy(locbuf, sndbuf, sizeof(sndbuf));
		}
		pthread_mutex_unlock(&mut);
		printf("CONSUMER2: %s\n", locbuf);
	}

	return NULL;
}

void *producer_threadfunc(void *par)
{
	int j = 0;
	while (true)
	{
		pthread_mutex_lock(&mut);
		{
			sprintf(sndbuf, "DATUM %d %10d", ++j, rand());
			printf("> Sending: %s\n", sndbuf);
			pthread_cond_broadcast(&cond);
		}
		pthread_mutex_unlock(&mut);

		sleep(2);
	}
	return NULL;
}

// In main():
 
	pthread_mutex_init(&mut, NULL);
	pthread_cond_init(&cond, NULL);

	pthread_create(&producer_th, NULL, producer_threadfunc, NULL);
	pthread_create(&consumer1_th, NULL, consumer1_threadfunc, NULL);
	pthread_create(&consumer2_th, NULL, consumer2_threadfunc, NULL);

	// main loop
	// ...

jeongchanKim, 2017-07-16:
I tested with your code, but it seems like working well.
If it is not correct what you expected, please explain your problem.

Thanks.

The below is the log for your test code.(I testet until DATUM 100)
-----------------------------------------------------------
TASH>>> Sending: DATUM 1          0
CONSUMER1 START
CONSUMER2 START
> Sending: DATUM 2      15407
CONSUMER1: DATUM 2      15407
CONSUMER2: DATUM 2      15407
> Sending: DATUM 3      13763
CONSUMER1: DATUM 3      13763
CONSUMER2: DATUM 3      13763
> Sending: DATUM 4      20812
CONSUMER1: DATUM 4      20812
CONSUMER2: DATUM 4      20812
> Sending: DATUM 5      31895
CONSUMER1: DATUM 5      31895
CONSUMER2: DATUM 5      31895
> Sending: DATUM 6      27184
CONSUMER1: DATUM 6      27184
CONSUMER2: DATUM 6      27184
> Sending: DATUM 7        973
CONSUMER1: DATUM 7        973
CONSUMER2: DATUM 7        973
> Sending: DATUM 8      15883
CONSUMER1: DATUM 8      15883
CONSUMER2: DATUM 8      15883
> Sending: DATUM 9      10054
CONSUMER1: DATUM 9      10054
CONSUMER2: DATUM 9      10054
> Sending: DATUM 10      27723
CONSUMER1: DATUM 10      27723
CONSUMER2: DATUM 10      27723
> Sending: DATUM 11       2096
CONSUMER1: DATUM 11       2096
CONSUMER2: DATUM 11       2096
> Sending: DATUM 12       5164
CONSUMER1: DATUM 12       5164
CONSUMER2: DATUM 12       5164
> Sending: DATUM 13      29622
CONSUMER1: DATUM 13      29622
CONSUMER2: DATUM 13      29622
> Sending: DATUM 14       5504
CONSUMER1: DATUM 14       5504
CONSUMER2: DATUM 14       5504
> Sending: DATUM 15      30406
CONSUMER1: DATUM 15      30406
CONSUMER2: DATUM 15      30406
> Sending: DATUM 16       1245
CONSUMER1: DATUM 16       1245

lucaing80, 2017-07-17:
Here's the log I get on Artik053 module running TizenRT with the exact example I provided you:

	U-Boot 2017.01-00065-g11a594a84763 (May 04 2017 - 21:26:44 +0900)

	CPU:   Exynos200 @ 320 MHz
	Model: ARTIK-053 based on Exynos T20
	DRAM:  722 KiB
	WARNING: Caches not enabled
	BL1 released at 2017-3-13 15:00
	Flash: 8 MiB
	*** Warning - bad CRC, using default environment

	In:    serial@80180000
	Out:   serial@80180000
	Err:   serial@80180000
	Hit any key to stop autoboot:  0
	gpio: pin gpg16 (gpio 46) value is 1
	## Starting application at 0x040C8020 ...
	s5j_sflash_init: FLASH Quad Enabled
	uart_register: Registering /dev/console
	uart_register: Registering /dev/ttyS0
	uart_register: Registering /dev/ttyS1
	uart_register: Registering /dev/ttyS2
	uart_register: Registering /dev/ttyS3
	uart_register: Registering /dev/ttyS4
	System Information:
			Version: 1.0
			Commit Hash: a847d0a5fe27ddb75f0b904b96c9ed8d0b101362
			Build User: samsung@samsung-OptiPlex-9020
			Build Time: 2017-05-11 11:17:20
			System Time: 07 Feb 2106, 06:28:15 [s] UTC Hardware RTC Support
	TASH>>START
	PRODUCER START
	> Sending: DATUM 1          0
	  sent
	CONSUMER1 START
	CONSUMER2 START

Then it's stuck indefinitely.
What version of TizenRT did you use? The HEAD of the Repo or the version released with ARTIK IDE?

Thank you.

jeongchanKim, 2017-07-18:
I tested with the latest code with artik053.
I modify the hello app with artik053/tash config.
I tested again, but it works well also...
(tested with this commit : commit 2da20e67ea50b027715eb0c6a0eaaa04c2c46708)

lucaing80, 2017-07-18:
Is there an official build I can test which does not exhibit the problem I'm experiencing? I don't want to rebuild TizenRT from sources.‎In any case I can provide you my e-mail (l.giuliani@gmail.com) so that you can contact me directly to remotely connect to my PC and see the problem live or to have Artik support center get in touch with me.                                                                                                                                     Thanks in advance,Luca                                                                                                                                                                                                   Inviato dallo smartphone BlackBerry 10.                                                                                                                                                                                                                Da: jeongchanKimInviato: martedì 18 luglio 2017 06:07A: Samsung/TizenRTRispondi a: Samsung/TizenRTCc: lucaing80; AuthorOggetto: Re: [Samsung/TizenRT] pthread_cond_broadcast + pthread_cond_wait BROKEN (#279)I tested with the latest code with artik053.
I modify the hello app with artik053/tash config.
I tested again, but it works well also...
(tested with this commit : commit 2da20e6)

  
  

 cript><br></div>

jeongchanKim, 2017-07-18:
Uhmm...
I don't know why you test with old version.
It is not easy for supporting your problem, because there is no issue on latest version.

sunghan-chang, 2017-07-31:
@lucaing80 I am sorry, but it is difficult to support old version. Do you have additional question? If you don't, let's close this.

sunghan-chang, 2017-08-07:
@lucaing80 There is no conversation for this. I am closing this. If you have additional question, please reopen. Thank you.

closed 2017-08-07 by sunghan-chang