Samsung/TizenRT #290: What is the automated way to build release|debug from a single config?

tadziopazur, 2017-07-17:
In nuttx, the flag is R: one wants to build release, they invoke:
nuttx/os$ make R=1
or 
nuttx/os$ make R=0

In TizenRT config system (Kconfig files, seen thru menuconfig make target) there are several flags, including:
DEBUG_SYMBOLS  DEBUG_FULLOPT  DEBUG_NOOPT  DEBUG_CUSTOM.
Some of them are not used in the build process (DEBUG_FULLOPT is not used at all).
The sources are built using ARCHOPTIMIZATION and MAXOPTIMIZATION, which are inconsistently set in Make.defs files (i.e.: some set MAXOPTIMIZATION to -O0).

How can one build TizenRT with consistent debug/release flags without changing sources (Make.defs)?

sunghan-chang, 2017-07-19:
We have a plan to support that functionality but not now.

sunghan-chang, 2017-08-09:
@tadziopazur I can not find R option on nuttx Makefile. Only V treated. Could you tell me where do you find it?

sunghan-chang, 2017-08-28:
@tadziopazur

closed 2017-08-28 by sunghan-chang