Samsung/TizenRT #316: GDB debugging fails for TizenRT/Artik053

pmarcinkiew, 2017-07-25:
I would like to debug TizenRT using GBD and OpenOCD. It fails for me. I compiled application demo and TASH is working over UART but debugging with GDB fails.

I started OpenOCD server using configuration from TizenRT repo:
```
$ tizenrt/build/configs/artik053/tools/openocd$ ./linux64/openocd -f artik053.cfg -d
```
When I connect using GDB:
```
(gdb) target remote localhost:3333
```

I get output:
```
#0  0x00000000 in ?? ()
```
All commands for GDB return errors or zeros.

My full GDB log:

```
$ arm-none-eabi-gdb tizenrt/build/output/bin/tinyara
GNU gdb (7.10-1ubuntu3+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from tizenrt/build/output/bin/tinyara...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb) 
```

My Full OpenOCD log:
```
$ tizenrt/build/configs/artik053/tools/openocd$ ./linux64/openocd -f artik053.cfg -d
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 1 command.c:544 command_print(): debug_level: 3
Debug: 14 1 options.c:181 add_default_dirs(): bindir=/home/heesub/Documents/tinyara/build/configs/artik053/tools/openocd/source/openocd/../openocd-install/bin
Debug: 15 1 options.c:182 add_default_dirs(): pkgdatadir=/home/heesub/Documents/tinyara/build/configs/artik053/tools/openocd/source/openocd/../openocd-install/share/openocd
Debug: 16 1 options.c:183 add_default_dirs(): exepath=/home/piotr/src/iotjs/deps/tizenrt/build/configs/artik053/tools/openocd/linux64
Debug: 17 1 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 1 configuration.c:42 add_script_search_dir(): adding /home/piotr/.openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/piotr/src/iotjs/deps/tizenrt/build/configs/artik053/tools/openocd/linux64/../share/openocd/site
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/piotr/src/iotjs/deps/tizenrt/build/configs/artik053/tools/openocd/linux64/../share/openocd/scripts
Debug: 21 1 configuration.c:82 find_file(): found artik053.cfg
Debug: 22 1 configuration.c:82 find_file(): found artik053-usb-ftdi.cfg
Debug: 23 1 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_interface ftdi
Debug: 24 1 command.c:143 script_debug(): command - interface ocd_interface ftdi
Debug: 26 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_device_desc'...
Debug: 27 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_serial'...
Debug: 28 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_location'...
Debug: 29 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_channel'...
Debug: 30 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_init'...
Debug: 31 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_layout_signal'...
Debug: 32 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_set_signal'...
Debug: 33 1 command.c:364 register_command_handler(): registering 'ocd_ftdi_get_signal'...
Debug: 34 2 command.c:364 register_command_handler(): registering 'ocd_ftdi_vid_pid'...
[... 170 lines cut ...]
Debug: 212 4 command.c:1098 help_add_command(): added 'mrw' help text
Debug: 213 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_add_help_text mrw Returns value of word in memory.
Debug: 214 4 command.c:143 script_debug(): command - add_help_text ocd_add_help_text mrw Returns value of word in memory.
Debug: 216 4 command.c:1111 help_add_command(): added 'mrw' help text
Debug: 217 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_add_usage_text mrb address
Debug: 218 4 command.c:143 script_debug(): command - add_usage_text ocd_add_usage_text mrb address
Debug: 220 4 command.c:1098 help_add_command(): added 'mrb' help text
Debug: 221 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_add_help_text mrb Returns value of byte in memory.
Debug: 222 4 command.c:143 script_debug(): command - add_help_text ocd_add_help_text mrb Returns value of byte in memory.
Debug: 224 4 command.c:1111 help_add_command(): added 'mrb' help text
Debug: 225 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_add_usage_text mmw address setbits clearbits
Debug: 226 4 command.c:143 script_debug(): command - add_usage_text ocd_add_usage_text mmw address setbits clearbits
Debug: 228 4 command.c:1098 help_add_command(): added 'mmw' help text
Debug: 229 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 230 4 command.c:143 script_debug(): command - add_help_text ocd_add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 232 4 command.c:1111 help_add_command(): added 'mmw' help text
Debug: 233 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_reset_config trst_and_srst srst_push_pull trst_push_pull
Debug: 234 4 command.c:143 script_debug(): command - reset_config ocd_reset_config trst_and_srst srst_push_pull trst_push_pull
User : 236 4 command.c:544 command_print(): trst_and_srst separate srst_gates_jtag trst_push_pull srst_push_pull connect_deassert_srst
Debug: 237 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_debug_level -1
Debug: 238 4 command.c:143 script_debug(): command - debug_level ocd_debug_level -1
debug_level: -1
undefined debug reason 7 - target needs reset

```

jhunahn, 2017-07-31:
Hello, pmarcinkiew.

I checked your logs. I think openocd failed to reset the device.
Would you like force reset the device using GDB?
following as:
```bash
(gdb) monitor reset halt
```

I expect the device to reset by JTAG. After reset the device, load the elf image agian.
(**You must enter the absolute path.**)
```bash
(gdb) load /home/jhun-ahn/workspace/a053/check/build/output/bin/tinyara
```

And then, try run.
```bash
(gdb) c
```

My GDB log is Follow as:
```bash
openocd git:(cm0) $ arm-none-eabi-gdb `pwd`/../../../../build/output/bin/tinyara
GNU gdb (7.10-1ubuntu3+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jhun-ahn/workspace/a053/check/build/output/bin/tinyara...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb) monitor reset halt
JTAG tap: s5jt200.dap tap/device found: 0x4ba00477 (mfg: 0x23b (ARM Ltd.), part: 0xba00, ver: 0x4)
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
JTAG-DP STICKY ERROR
s5jt200.r4: ran after reset and before halt ...
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x600001d3 pc: 0x00000164
D-Cache: disabled, I-Cache: disabled
(gdb) load /home/jhun-ahn/workspace/a053/check/build/output/bin
"/home/jhun-ahn/workspace/a053/check/build/output/bin" is not an object file: Is a directory
(gdb) load /home/jhun-ahn/workspace/a053/check/build/output/bin/tinyara
Loading section .header, size 0x20 lma 0x40c8000
Loading section .text, size 0xbfe26 lma 0x40c8020
Loading section .ARM.exidx, size 0x8 lma 0x4187e48
Loading section .data, size 0x12e8 lma 0x4187e50
Loading section .padding, size 0x4c8 lma 0x4189138
Start address 0x40c8060, load size 792062
Transfer rate: 54 KB/sec, 4020 bytes/write.
(gdb) c
Continuing.
```

Your feedback would be greatly appreciated.
Best Regards
Jihun

sunghan-chang, 2017-08-07:
@pmarcinkiew Is this enough answer for you? If it is, let's close this.

marcinsmsg, 2017-08-22:
I've tried to reset Artik using hard reset and load symbols. However, gdb is not able to resolve any symbols.

OpenOCD log:
```
debug_level: -1

undefined debug reason 7 - target needs reset
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x040cb874
D-Cache: enabled, I-Cache: enabled
```
GDB log:
```
(gdb) target remote localhost:3333
A program is being debugged already.  Kill it? (y or n) y

Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb) monitor reset halt
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x040cb874
D-Cache: enabled, I-Cache: enabled
(gdb) load /home/m.ziombski/pr2/tizenrt/build/output/bin/tinyara
Loading section .header, size 0x20 lma 0x40c8000
Loading section .text, size 0x126b82 lma 0x40c8020
Loading section .ARM.exidx, size 0x8 lma 0x41eeba4
Loading section .data, size 0x1370 lma 0x41eebac
Loading section .padding, size 0x4e4 lma 0x41eff1c
Start address 0x40c8060, load size 1213438
Transfer rate: 54 KB/sec, 4044 bytes/write.
(gdb) b i2c_uioregister
No symbol table is loaded.  Use the "file" command.
Make breakpoint pending on future shared library load? (y or [n]) n        
(gdb) info registers
r0             0x0	0
r1             0x0	0
r2             0x0	0
r3             0x0	0
r4             0x0	0
r5             0x0	0
r6             0x0	0
r7             0x0	0
r8             0x0	0
r9             0x0	0
r10            0x0	0
r11            0x0	0
r12            0x0	0
sp             0x0	0x0
lr             0x0	0x0
pc             0x40c8060	0x40c8060
cpsr           0x0	0
(gdb) continue
Continuing.
^C 
Program received signal SIGINT, Interrupt.
0x040cb878 in ?? ()
(gdb) b i2c_uioregister
No symbol table is loaded.  Use the "file" command.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) b iotjs_main
No symbol table is loaded.  Use the "file" command.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) info functions
All defined functions:
(gdb) info registers
r0             0x0	0
r1             0x0	0
r2             0x0	0
r3             0x0	0
r4             0x2025404	33707012
r5             0x2025404	33707012
r6             0x20254d8	33707224
r7             0x2	2
r8             0x20c6884	34367620
r9             0x2044ef0	33836784
r10            0x0	0
r11            0x0	0
r12            0x1	1
sp             0x206e600	0x206e600
lr             0x40cb878	0x40cb878
pc             0x40cb878	0x40cb878
cpsr           0x60000153	1610613075
(gdb) 
```

pmarcinkiew, 2017-08-22:
I tried to load symbols from tinyara binary into separate instance of GDB:
```
arm-none-eabi-gdb tinyara
```
I can load symbols into one instance of GDB and connect to OpenOCD with another one, but still I can't setup working break point. When I try to use stepi to debug what happen when iotjs command is executed I fail to have break point triggered. 

It is clear that any valid debugging is still not possible with the latest TizenRT. I requested your help more than two weeks ago and still you fail to provide any valid debugging guide for Tizen RT at Artik 053.

Would you be so kind to provide guide how debugging of this OS should be conducted?

Please find below more detailed logs from my debugging.

I can setup break point for iotjs_main function:

```
piotr@bdell:~/src/iotjs/deps/tizenrt/build/output/bin$ arm-none-eabi-gdb tinyara
GNU gdb (7.6.50.20131218-0ubuntu1+1) 7.6.50.20131218-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from tinyara...done.
(gdb) bt iotjs_main
No stack.
(gdb) b iotjs_main
Breakpoint 1 at 0x40e17ac: file iotjs_main.c, line 161.
(gdb) 
```

I also can decode address from GDB connected to OpenOCD and setup break points but they are not triggered, when iotjs_main is executed.
```
piotr@bdell:~$ arm-none-eabi-gdb 
GNU gdb (7.6.50.20131218-0ubuntu1+1) 7.6.50.20131218-cvs
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00000000 in ?? ()
(gdb)  monitor reset halt
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x040ce19c                                                                                                                                                                                  
D-Cache: enabled, I-Cache: enabled                                                                                                                                                                               
(gdb) load /home/piotr/src/iotjs/deps/tizenrt/build/output/bin/tinyara
Loading section .header, size 0x20 lma 0x40c8000                                                                                                                                                                 
Loading section .text, size 0x1c4ea0 lma 0x40c8020                                                                                                                                                               
Loading section .ARM.exidx, size 0x8 lma 0x428cec0                                                                                                                                                               
Loading section .data, size 0x147c lma 0x428cec8                                                                                                                                                                 
Loading section .padding, size 0x4bc lma 0x428e344                                                                                                                                                               
Start address 0x40c8060, load size 1861632                                                                                                                                                                       
Transfer rate: 54 KB/sec, 4064 bytes/write.                                                                                                                                                                      
[... 177 lines cut ...]
0x040cb3fc in ?? ()
(gdb) stepi
0x040cb400 in ?? ()
(gdb) stepi
0x040cb404 in ?? ()
(gdb) stepi
0x040cb408 in ?? ()
(gdb) stepi
0x040cb40c in ?? ()
(gdb) stepi
0x040cb410 in ?? ()
(gdb) stepi
0x040ce34c in ?? ()
(gdb) stepi
0x040ce350 in ?? ()
(gdb) stepi
0x040ce354 in ?? ()
(gdb) stepi
0x040ce358 in ?? ()
(gdb) stepi
0x040ce35c in ?? ()
(gdb) stepi
0x040ce360 in ?? ()
(gdb) 
```

jhunahn, 2017-08-24:
Hello, pmarcinkiew.

Did you include debug symbol when build tizenrt?
You can check it as following as:

```bash
tizenrt/os $ size -A ../build/output/bin/tinyara
../build/output/bin/tinyara  :
section              size       addr
.header                32   67928064
.text              135020   67928096
.ARM.exidx              8   68063116
.data                1858   33699840
.bss                 7968   33701704
.comment               56          0
.ARM.attributes        51          0
.debug_abbrev      155776          0
.debug_info        814083          0
.debug_line        185176          0
.debug_aranges      15768          0
.padding             1322   68064982
.debug_loc         158596          0
.debug_ranges       12864          0
.debug_str          38502          0
.debug_frame        28212          0
Total             1555292
```
If you can see each section of ".debug*", the binary already include debug symbols.
But, in the other case, you should be enable the debug symbols using menuconfig.
```bash
tizenrt/os $ make menuconfig
```

Please try to build tizenrt after select debug option.

```
Build Setup > Debug Options > Generate Debug Symbols
```

finally, my logs is fallowing as:

1. execute of openocd
```bash
tizenrt/os $ openocd -f artik053.cfg -s ../build/configs/artik053/tools/openocd -c 'init;reset halt'
Open On-Chip Debugger 0.10.0+dev-00172-g7719e96-dirty (2017-08-25-10:08)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
force hard breakpoints
trst_and_srst separate srst_gates_jtag trst_push_pull srst_push_pull connect_deassert_srst
adapter_nsrst_assert_width: 50
adapter_nsrst_delay: 100
debug_level: -1
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x600001d3 pc: 0x00000164
D-Cache: disabled, I-Cache: disabled
```

2. my GDB logs
```bash
tizenrt/os $ arm-none-eabi-gdb /home/jhun-ahn/workspace/a053/check/build/output/bin/tinyara
\GNU gdb (7.10-1ubuntu3+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jhun-ahn/workspace/a053/check/build/output/bin/tinyara...done.
(gdb) target remote :3333
Remote debugging using :3333
0x00000164 in ?? ()
(gdb) load
Loading section .header, size 0x20 lma 0x40c8000
Loading section .text, size 0x20f6c lma 0x40c8020
Loading section .ARM.exidx, size 0x8 lma 0x40e8f8c
Loading section .data, size 0x742 lma 0x40e8f94
Loading section .padding, size 0x52a lma 0x40e96d6
Start address 0x40c8060, load size 138240
Transfer rate: 54 KB/sec, 3736 bytes/write.
(gdb) b i2c_uioregister
Breakpoint 1 at 0x40dee00: file i2c/i2c_uio.c, line 273.
(gdb) info registers
r0             0x80180000	2149056512
r1             0x4	4
r2             0x0	0
r3             0x4	4
r4             0x3	3
r5             0x1	1
r6             0x2025450	33707088
r7             0x20a882a2	547914402
r8             0x2020140	33685824
r9             0x80000000	2147483648
r10            0x0	0
r11            0x0	0
r12            0x0	0
sp             0x20210f0	0x20210f0
lr             0x1bbc	0x1bbc
pc             0x40c8060	0x40c8060 <__start>
cpsr           0x600001d3	1610613203
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x040cb834 in thread_create (name=0x202537c <g_irqvector+572> "", ttype=<optimized out>, priority=547914402, 
    stack_size=33685824, entry=0xdead, argv=0x0) at task/task_create.c:185
185		if (ret < 0) {
(gdb) 
```

I hope my comments helped you.
Thank you

pmarcinkiew, 2017-09-04:
Thank you very much for your comments.

I'm afraid that still your are missing key element in my questions. I need to set break point to analyze one function called for my project. You fail to obtain any working breakpoint trigger in your log.

Let's start with your logging session with breakpoint for i2c_uioregister.

```
^C
Program received signal SIGINT, Interrupt.
os_start () at init/os_start.c:480
480                     pthread_initialize();
(gdb) b tash_help
Breakpoint 2 at 0x40f32fc: file tash_command.c, line 146.
(gdb) c
Continuing.
```
I enabled breakpoint for help in tash and next I type help in TASH:

```
TASH>>help
         TASH command list 
         --------------------
cat              cd               date             df               
dhcpd            exit             free             getenv           
heapinfo         hello            help             ifconfig         
ifdown           ifup             iperf            kill             
killall          logm             ls               mkdir            
mkrd             mksmartfs        mount            ping             
ps               pwd              ramtest          reboot           
rm               rmdir            setenv           sh               
sleep            stkmon           umount           unsetenv         
uptime           wifi             wifi_scan        
TASH>>

```
It is clear that help is displayed so tash_help is called.c:

```
142 static int tash_help(int argc, char **args)                                                                                                                            
143 {                                                                                                                                                                     
144     int cmd_idx;                                                                                                                                                      
145                                                                                                                                                                       
146     printf("\t TASH command list \n");                                                                                                                                
147     printf("\t --------------------\n");                                                                                                                              
148    
```

The key issue here is that at GDB there is no any break point activated for this event. Would you be so kind to try to debug with breakpoint active?

tosyu, 2017-09-04:
Hello, I have tried gdb on tizenrt, breakpoint operation worked for me, but I had encountered problems with continue and next operations.

See attached logs:

```bash
GNU gdb (7.10-1ubuntu3+9) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ../build/output/bin/tinyara...done.
(gdb) target remote :3333
Remote debugging using :3333
0x02022530 in ?? ()
(gdb) load
Loading section .header, size 0x20 lma 0x40c8000
Loading section .text, size 0x1c59da lma 0x40c8020
Loading section .ARM.exidx, size 0x8 lma 0x428d9fc
Loading section .data, size 0x148c lma 0x428da04
Loading section .padding, size 0x570 lma 0x428ee90
Start address 0x40c8060, load size 1864702
Transfer rate: 52 KB/sec, 4062 bytes/write.
(gdb) b i2c_uioregister
Breakpoint 1 at 0x41c81c8: file i2c/i2c_uio.c, line 274.
(gdb) info registers
r0             0x800c1460	2148275296
r1             0x4	4
r2             0x2000	8192
r3             0x2000	8192
r4             0x2020000	33685504
r5             0x0	0
r6             0x0	0
r7             0x2	2
r8             0x2020140	33685824
r9             0x80000000	2147483648
r10            0x0	0
r11            0x0	0
r12            0x1	1
sp             0x20210d0	0x20210d0
lr             0x202265c	0x202265c
pc             0x40c8060	0x40c8060 <__start>
cpsr           0x600001d3	1610613203
(gdb) c
Continuing.

Breakpoint 1, i2c_uioregister (path=0x206f798 "/dev/i2c-0", dev=0x2023a80 <s5j_i2c0_priv>)
    at i2c/i2c_uio.c:274
274		dbg("Registering %s\n", path);
(gdb) c
Continuing.

```

Breakpoint is triggered but not once like its suppose to be. Each continue operation triggers the same breakepoint over and over again. There is no new input from console

```bash

Breakpoint 1, i2c_uioregister (path=0x206f798 "/dev/i2c-0", dev=0x2023a80 <s5j_i2c0_priv>)
    at i2c/i2c_uio.c:274
274		dbg("Registering %s\n", path);
(gdb) c
Continuing.

```

The same things happens for tash_help

```bash
^C
Program received signal SIGINT, Interrupt.
os_start () at init/os_start.c:588
588			up_idle();
(gdb) b tash_help
Breakpoint 2 at 0x40e2300: file tash_command.c, line 146.
(gdb) c
Continuing.

Breakpoint 2, tash_help (argc=1, args=0x2088f1c) at tash_command.c:146
146		printf("\t TASH command list \n");
(gdb) c
Continuing.

```

The same thing happens here too.

```bash

Breakpoint 2, tash_help (argc=1, args=0x2088f1c) at tash_command.c:146
146		printf("\t TASH command list \n");
(gdb) c
Continuing.

```

Next operation does jump to unknown symbol.

```bash
Breakpoint 2, tash_help (argc=1, args=0x2088f1c) at tash_command.c:146
146		printf("\t TASH command list \n");
(gdb) n
0xffff0018 in ?? ()
(gdb) n
Cannot find bounds of current function
```

List operation looks to work.

```bash
(gdb) l
141	 */
142	static int tash_help(int argc, char **args)
143	{
144		int cmd_idx;
145	
146		printf("\t TASH command list \n");
147		printf("\t --------------------\n");
148	
149		if (is_sorted_list == FALSE) {
150			tash_sort_cmds();
```

I'm using openocde:

```bash
Open On-Chip Debugger 0.10.0
```

olegartys, 2017-09-05:
@tosyu Hello, please try to disable ISR: 
`(gdb) monitor cortex_r4 maskisr on`

closed 2019-04-03 by sunghan-chang