Samsung/TizenRT #317: 'date' command does not work

JunhwanPark, 2017-07-26:
Hello.

When I enable 'CONFIG_LIBC_LOCALTIME' option in menuconfig, 'date' command does not work.
I am trying to change the date and time using 'date -s'. However, it is always fixed at 1970. It must be disabled by 'CONFIG_LIBC_LOCALTIME'. Is this normal?

junmin-kim, 2017-08-03:
I found that there are some issues in timezone setting.
After I fix these problem, I'll inform to you.

JunhwanPark, 2017-08-03:
@junmin-kim 
I'll wait for your patch.
Thank you.

junmin-kim, 2017-08-23:
We merged this commit #365

JunhwanPark, 2017-08-23:
Hi @junmin-kim , @thapav ,
Thanks for implementing it. I updated to the latest source.
I set it to `artik053/nettest` and enabled only `CONFIG_LIBC_LOCALTIME`. And I tested it like this:
```
TASH>> date -s Apr 21 10:35:22 1991 Sun 175 1
```
But I did not get the result I wanted. 
```
TASH>> date
Jan 01 00:00:00 1970 Thu
```
And the user must set all day of week, day of year, and daylight saving time? 
This is so confusing.

JunhwanPark, 2017-08-23:
I tested it simply. How about calculating this in the kernel? I want to know your thoughts.

```c
static int day_of_week(int y, int m, int d)
{
	static int t[] = {0, 3, 2, 5, 0, 3, 5, 1, 4, 6, 2, 4};

	y -= m < 3;

	return (y + y/4 - y/100 + y/400 + t[m - 1] + d) % 7;
}

static int day_of_year(int y, int m, int d)
{
	static int month[] = {31, 28, 31, 30, 31, 30, 31, 31, 30, 31, 30, 31};
	int i;
	long total = 0L;

	total = (y - 1) * 365 + (y - 1) / 4 - (y - 1) / 100 + (y - 1) / 400;
	if ((y % 4 == 0) && ((y % 100 != 0) || (y % 400 == 0))) {
		month[1]++;
	}
	for (i = 0; i < m - 1; i++) {
		total += month[i];
	}
	total += d;

	return total;
}

/* ... */

dow = day_of_week(y + 1900, m + 1, d);
doy = day_of_year(y + 1900, m + 1, d) - day_of_year(y + 1900, 1, 1);
```

JunhwanPark, 2017-08-24:
@thapav 
I tested it with 'usage'. Is it wrong? Let me know the exact usage.

JunhwanPark, 2017-08-24:
@thapav 
I have tested it as follows. In my opinion, you only need to change 'usage'. I'll close this issue.
```
date -s Apr 21 10:35:22 1991 Sun
date -s Apr 21 10:35:22 1991 Sun 0
date -s Apr 21 10:35:22 1991 Sun 175 0
```

closed 2017-08-24 by JunhwanPark