Samsung/TizenRT #322: I2C read does not work on Artik053

tadziopazur, 2017-07-27:
I have some questions regarding I2C implementation on Artik.

1. Why does read fail to generate a correct I2C transaction? What can I do to fix it?
2. The i2c driver does not finish the transaction when there is a write of 1 (one) byte. Why?
How are devices that do not have registers (i.e.: pcf8574) supposed to be handled?
3. The I2C signals driven by artik053 devkit rise slowly. Is it relevant to the other observed behavior?

Details follow.

First, the i2c_read did not work for me: artik failed to put a transaction onto the bus (not even START has been generated, just three stray SCL pulses -- please note SDA is high all the time). Scope diagram:
[image]

Write works somewhat better, in the sense that there are correct transactions (with an exception) issued onto the bus:
[image]

This is what my test suite looks like:
[image]

BTW artik devboard I2C output rises slowly, at about 200ns (at 400kHz bus clock) to reach TTL 1. I wonder if this is of any relevance (scope diagram attached):
[image]

pmarcinkiew, 2017-07-30:
Please double check if your probe compensation is valid.

[image]

See Tektronix manual for reference:
https://neurophysics.ucsd.edu/Manuals/Tektronix/TDS%201000B%20and%20TDS%202000B%20Manual.pdf

tadziopazur, 2017-07-30:
The probe is rated at 200MHz, 10MOhm, <12pF. I checked it at the 1kHz supplied by the scope, it looked good.
I think the signal (400kHz square waveform) is within the operating range of the scope (+probes).
Still I am not sure why these slopes rise so slowly, nor do I know if that affects the I2C communication.
It does not look like the communication is affected (at least not with Arduino Uno on the other side).

rzr, 2018-03-07:
i2c worked for me on ARTIK055S (using IoT.js) I can share more details if it helps.

rzr, 2018-08-09:
Relate-to: https://github.com/Samsung/TizenRT/issues/1001

rzr, 2018-11-15:
Does i2ctool detect any item on bus ?

You need to rebase on it to use it:
https://github.com/Samsung/TizenRT/pull/2439#event-1959436602

closed 2019-11-26 by sunghan-chang