Samsung/TizenRT #329: Does tizenrt support command "cpuinfo", or some other command such as "cpuinfo"?

cooldo, 2017-08-01:


cooldo, 2017-08-01:
I want to get the cpu information by command line.
How can I get this information?
Thank you very much.
Your response will be deeply thanks.

jeongarmy, 2017-08-01:
I don't know what cpu information means exactly?
you mean /proc/cpuinfo which show somethings like model name, processor in linux?
Please let me know what you want to know using this command exactly and why.

cooldo, 2017-08-01:
Thank you very much for your response.   
We want to use the command such as linux "top", to print the cpu usage of each process. 
we are designing "system log" IDE,  our purpose is to capture the system log from "tash", and display into our IDE graphicly.
Thank you very much.

jeongarmy, 2017-08-01:
I'm sorry that we don't support that function now.
But you can get them from proc file system.
You can get cpu usage of each process by reading /proc/[pid]/loadavg.
We will consider whether to support command like top in linux or not.

cooldo, 2017-08-01:
Thank you very much, it is very helpful.

closed 2017-08-01 by cooldo