Samsung/TizenRT #344: when I 'LIB_SYSCALL' enable, meet build error.

JunhwanPark, 2017-08-03:
I downloaded the latest version.
If you enable 'System Call' in menuconfig and compile, a build error occurs.
(Same as enabling CONFIG_LIBC_SYSCALL).

```bash
os/include/unistd.h: In function '_exit':
proxies/PROXY__exit.c:10:1: error: 'noreturn' function does return [-Werror]
 }
 ^
cc1: all warnings being treated as errors
Makefile:92: recipe for target 'PROXY__exit.o' failed
make[1]: *** [PROXY__exit.o] Error 1
```

sunghan-chang, 2017-08-07:
@JunhwanPark I am sorry for delay. Do you enable protected build also? Without protected build, syscall is not needed.

JunhwanPark, 2017-08-07:
@sunghan-chang OK. I also enable CONFIG_BUILD_PROTECTED. But I met also build error. Does TizenRT officially support 'Protect Mode'?

sunghan-chang, 2017-08-09:
@JunhwanPark Not yet. To use protected build, we should add more files, like up_userspace.c.

sunghan-chang, 2017-08-09:
@JunhwanPark It worked well before. But after merging DS pull reqeust, some files are removed and some setting like MPU was changed. So, it is not supported now.

JunhwanPark, 2017-08-09:
@sunghan-chang 
Thank you for reply.

Really??
I reset initial version.

```bash
$ git hist 
* 4fc385d 2016-12-23 | The first commit of Tizen RT 1.0 M1 (HEAD -> dev) [junmin.kim]
* ad287a1 2016-08-30 | Initial empty repository [Jong Jeong Kim]
```
And, I built it. But, also I met compile error.
```bash
$ cd os/tools
$ sh configure.sh sidk_s5jt200/hello_with_tash
$ cd ..
$ make menuconfig
 > Enable 'System Call'
$ make
```
Can you tell me when I need to check the version?

sunghan-chang, 2017-08-09:
@JunhwanPark Internally we had, not in here.
Anyway currently we don't support the protected build so that we don't need to use syscall either.

JunhwanPark, 2017-08-09:
@sunghan-chang OK. I see.
Thanks for information.

closed 2017-08-09 by JunhwanPark

sunghan-chang, 2017-09-01:
@JunhwanPark
1. We modified this. When you enable PROTECTED_BUILD, you can find syscall menu. Before enabling it, you can't find syscall on menuconfig. 
2. PROTECTED_BUILD is under developing, but because of some reason, that is delayed. Anyway we will add it.

JunhwanPark, 2017-09-01:
@sunghan-chang  Thanks. I will try.