Samsung/TizenRT #346: Boot fails for artik053/tash compiled with GCC 5.4

pmarcinkiew, 2017-08-03:
When I compile TizenRT with config artik053/tash it fails during boot with abort.

This is my boot log
```
U-Boot 2017.01-00065-g11a594a84763 (May 04 2017 - 21:26:44 +0900)

CPU:   Exynos200 @ 320 MHz
Model: ARTIK-053 based on Exynos T20
DRAM:  722 KiB
WARNING: Caches not enabled
BL1 released at 2017-3-13 15:00
Flash: 8 MiB
*** Warning - bad CRC, using default environment

In:    serial@80180000
Out:   serial@80180000
Err:   serial@80180000
Hit any key to stop autoboot:  0 
gpio: pin gpg16 (gpio 46) value is 1
## Starting application at 0x040C8020 ...
s5j_sflash_init: FLASH Quad Enabled
arm_dataabort: 
Data abort. PC: 040c861c DFAR: fffffffc DFSR: 0000080d
up_assert: Assertion failed at file:armv7-r/arm_dataabort.c line: 111 task: �
```

pmarcinkiew, 2017-08-03:
Fix prepared by Jarosław Pelczar: https://github.com/Samsung/TizenRT/pull/347

sunghan-chang, 2017-08-07:
@pmarcinkiew Thank you for contibuting. Your pull request was merged. Let's close this.

closed 2017-08-10 by sunghan-chang