Samsung/TizenRT #348: Missing tinyara headers

rzr, 2017-08-03:
Hi,
I just had a quick look at TizenRT, and noticed that some included headers are missing, either they should be added or codebase should be cleaned.

Here a list of non available included headers:

```
tinyara/analog/dac.h
tinyara/binfmt/elf.h
tinyara/drivers/rwbuffer.h
tinyara/errno.h
tinyara/fs/automount.h
tinyara/input/kbd_codec.h
tinyara/input/mouse.h
tinyara/input/touchscreen.h
tinyara/lcd/lcd.h
tinyara/module.h
tinyara/mtd/mtd.h
tinyara/net/arp.h
tinyara/net/igmp.h
tinyara/net/pkt.h
tinyara/net/tcp.h
tinyara/net/udp.h
tinyara/scsi.h
tinyara/sensors/ppd42ns.h
```

Some are in TizenRT/os/include/ (ie: errno.h) while others are just gone (ie: automount.h) also some symbols are unresolved (ie: arp_out).

Any hints are welcome.

sunghan-chang, 2017-08-07:
@rzr Thank you for notice.
We will clean them.

sunghan-chang, 2017-08-10:
@rzr All of header files which we can modify are fixed by #367, #364 and #359.
Some are removed, some are added. Do you have any query?

rzr, 2017-08-10:
I can also remove lcd part until it is fixed,  and then I will make a check and close this bug would this help?

sunghan-chang, 2017-08-10:
sure.

sunghan-chang, 2017-08-28:
Hello @rzr .
Is there any additional change?

rzr, 2017-08-28:
can I also remove lcd code like sensor_test ? 

'''
./os/syscall/syscall_funclookup.c:#include <tinyara/errno.h>
./os/drivers/Kconfig:           drivers support interfaces as defined in include/tinyara/lcd/lcd.h
./os/drivers/lcd/ili9341.c:#include <tinyara/lcd/lcd.h>
'''

sunghan-chang, 2017-08-30:
Ok, let's remove lcd part on drvier and include. But leave errno.h at syscall.
Do you have a plan for this? or Do we?

sunghan-chang, 2017-09-20:
@rzr There is no activity related this. We will remove lcd part later. I close this, if you have another reqeust or question, please reopen this.

closed 2017-09-20 by sunghan-chang

rzr, 2017-09-27:
ok for me , I am away this month I will open new PR if I can see other improvement , thanks again