Samsung/TizenRT #423: Which Kconfig has the entry of "NET_LWIP_DHCPC" ?

manoharahk, 2017-08-17:
https://github.com/Samsung/TizenRT/blob/d8b91b34e2ce2fc8b42644806c27bb39c2446ac2/apps/examples/wakaama_client/wakaama_client.c#L45

sinzah, 2017-08-18:
As I knew, there are no entry to configure NET_LWIP_DHCPC.
CONFIG_NETUTILS_DHCPC is used to dhcp client instead lwip dhcp

pillip8282, 2017-08-21:
Yes he is right. TizenRT doesn't use CONFIG_NETUTILS_DHCPC anymore.

sunghan-chang, 2017-09-20:
@manoharahk Do you have additional question? If you don't, let's close this.

manoharahk, 2017-10-10:
OK, then are we creating missing entry to configure?

sunghan-chang, 2017-10-12:
@manoharahk Just removing that line is enough. There is no config NET_LWIP_DHCPC and we will not use that one.

manoharahk, 2017-10-16:
OK. Thanks you all for your responses on this issue.

closed 2017-10-16 by manoharahk