Samsung/TizenRT #441: Flashin with make download ALL fails for Artik 053

pmarcinkiew, 2017-08-22:
I compiled TizenRT for Artik053 at fresh installed Ubuntu with dependencies installed. It fails to flash Artik053 board with OpenOCD error when make download ALL is executed.

This log is very similar to issue described at artil.io:

https://developer.artik.io/forums/t/how-to-do-onboarding-with-artik053-board/3122

There is not solution described in artik.io or any mention is Tizen RT source code how to solve this issue.

Flash log:
```
piotr@bdell:~/src/iotjs/deps/tizenrt/os$ make download ALL
openocd is picked from /home/piotr/src/iotjs/deps/tizenrt/os/../build/configs/artik053/tools/openocd/linux64
Binaries are picked from /home/piotr/src/iotjs/deps/tizenrt/os/../build/output/bin
Board path is /home/piotr/src/iotjs/deps/tizenrt/os/../build/configs/artik053
ALL :
~/src/iotjs/deps/tizenrt/build/configs/artik053/tools/openocd ~/src/iotjs/deps/tizenrt/os
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 2000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
force hard breakpoints
trst_and_srst separate srst_gates_jtag trst_push_pull srst_push_pull connect_deassert_srst
debug_level: -1
```

pmarcinkiew, 2017-08-22:
I found usb configuration file for Artik 053, which works for my Ubuntu:

```
piotr@bdell:~$ cat /etc/udev/rules.d/99-bcm-artik05x.rules 
SUBSYSTEMS=="usb",ATTRS{idVendor}=="0a5c",ATTRS{idProduct}=="43fa",MODE="0666" RUN+="/sbin/modprobe ftdi_sio" RUN+="/bin/sh -c 'echo 0a5c 43fa > /sys/bus/usb-serial/drivers/ftdi_sio/new_id'"
SUBSYSTEMS=="usb",ATTRS{idVendor}=="0403",ATTRS{idProduct}=="6010",MODE="0666" RUN+="/sbin/modprobe ftdi_sio" RUN+="/bin/sh -c 'echo 0403 6010 > /sys/bus/usb-serial/drivers/ftdi_sio/new_id'"
```

Please consider extending your very good and helpful documentation with one more details about this configuration issue.

sunghan-chang, 2017-08-23:
@pmarcinkiew Thank you for giving opinion.
Even though we have this internally, because this is optional, we didn't add this at document.
As you said, this can be helpful. So added #442 .

sunghan-chang, 2017-08-28:
@pmarcinkiew If you don't have any additional query, I will close this at wednesday.

closed 2017-08-30 by sunghan-chang