Samsung/TizenRT #451: Artik Demo Example LD Error

iamsanjeev, 2017-08-28:
configure.sh artik053/iotivity
make menuconfig -> choose "Artik Demo Examples from Application Configuration Menu"

LD: tinyara
$HOME/tizenrt/TizenRT/os/../build/output/libraries/libapps.a(artik_demo.o): In function `wifiAutoConnectInit':
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:273: undefined reference to `dm_conn_register_linkup_cb'
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:277: undefined reference to `dm_conn_get_interface'
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:282: undefined reference to `WifiIsConnected'
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:298: undefined reference to `get_security_config'
$HOME/tizenrt/TizenRT/os/../build/output/libraries/libapps.a(artik_demo.o): In function `artik_demo_run':
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:910: undefined reference to `dm_conn_get_rssi'
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:915: undefined reference to `dm_conn_get_tx_power'
$HOME/tizenrt/TizenRT/apps/examples/artik_demo/artik_demo.c:920: undefined reference to `dm_conn_get_channel'

sunghan-chang, 2017-08-28:
@sangwon03 @an4967 @pillip8282 
There are two problems on this issue.
1. This example uses dm functionality but there is no dependancy.
2. Some wifi function names are changed but not in this example.

sunghan-chang, 2017-08-30:
#452 was merged so that I close this.

closed 2017-08-30 by sunghan-chang