Samsung/TizenRT #546: Buildbreak at TizenRT HEAD

tadziopazur, 2017-09-07:
As of commit 3f599c21b63cedcf93f78296916948bb2aa7b395  TizenRT cannot be built for iotjs (do you need the defconfig?).
There might be more issues I didn't see with my brief examination.

Option CONFIG_EXAMPLES_SLSIWIFI enabled causes:
> CC:  nettest_functions.c
>nettest_functions.c:29:37: fatal error: slsi_wifi/slsi_wifi_api.h: No such file or directory
> #include <slsi_wifi/slsi_wifi_api.h>
>                                     ^
>compilation terminated.
>Makefile:86: recipe for target 'nettest_functions.o' failed
>make[2]: *** [nettest_functions.o] Error 1
>make[2]: Leaving directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/apps/examples/slsiwifi'
>Makefile:109: recipe for target 'examples/slsiwifi_all' failed

Option CONFIG_NET_CMDS enabled causes:
>make[2]: Entering directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/apps/system/utils'
>./netcmd.c:49:29: fatal error: protocols/dhcpc.h: No such file or directory
> #include <protocols/dhcpc.h>
>                             ^
>compilation terminated.
>ERROR: arm-none-eabi-gcc failed: 1
>       command: arm-none-eabi-gcc -M -fno-builtin -mcpu=cortex-r4 -mfpu=vfpv3 -Wall -Werror -Wstrict-prototypes -Wshadow -Wundef -Wno-implicit-function-declaration -Wno-unused-function -Wno-unused-but-set-variable -g -O0 -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -I. -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/include -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/../framework/include -D__KERNEL__ -pipe -ffunction-sections -fdata-sections -I../../../external/dhcpd/ ./netcmd.c
>Makefile:222: recipe for target '.depend' failed
>make[2]: *** [.depend] Error 1
>make[2]: Leaving directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/apps/system/utils'

Options CONFIG_DM combined with CONFIG_ARCH_CHIP_S5JT200 enabled cause:
>make[1]: Entering directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/framework'
>src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c:43:29: fatal error: protocols/dhcpc.h: No such file or directory
> #include <protocols/dhcpc.h>
>                             ^
>compilation terminated.
>ERROR: arm-none-eabi-gcc failed: 1
>       command: arm-none-eabi-gcc -M -fno-builtin -mcpu=cortex-r4 -mfpu=vfpv3 -Wall -Werror -Wstrict-prototypes -Wshadow -Wundef -Wno-implicit-function-declaration -Wno-unused-function -Wno-unused-but-set-variable -g -O0 -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -I. -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/include -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/../framework/include -D__KERNEL__ -pipe -ffunction-sections -fdata-sections -I/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/../framework/include/iotbus -I/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/../framework/include/dm src/dm/arch/sidk_s5jt200/s5j_dm_connectivity.c
>Makefile:69: recipe for target '.depend' failed
>make[1]: *** [.depend] Error 1
>make[1]: Leaving directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/framework'

External mbedtls does not build as well (CONFIG_NET_SECURITY_TLS enabled):
>make[2]: Entering directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/external/mbedtls'
>./aes.c:46:28: fatal error: mbedtls/config.h: No such file or directory
> #include "mbedtls/config.h"
>                            ^
>compilation terminated.
>ERROR: arm-none-eabi-gcc failed: 1
>       command: arm-none-eabi-gcc -M -fno-builtin -mcpu=cortex-r4 -mfpu=vfpv3 -Wall -Werror -Wstrict-prototypes -Wshadow -Wundef -Wno-implicit-function-declaration -Wno-unused-function -Wno-unused-but-set-variable -g -O0 -fno-strict-aliasing -fno-strength-reduce -fomit-frame-pointer -I. -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/include -isystem /opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/os/../framework/include -D__KERNEL__ -pipe -ffunction-sections -fdata-sections ./aes.c
>Makefile:131: recipe for target '.depend' failed
>make[2]: *** [.depend] Error 1
>make[2]: Leaving directory '/opt/iotjs/iotjs-nuttx/iotjs/deps/tizenrt/external/mbedtls'
>Makefile:112: recipe for target 'mbedtls_depend' failed
>make[1]: *** [mbedtls_depend] Error 2

btheosam, 2017-09-07:
https://github.com/Samsung/TizenRT/pull/543 had caused build break for all commits after. ONce that was removed, the build was okay.

tadziopazur, 2017-09-07:
If it is that commit that break HEAD and 'removing' it (git revert --help) is acceptable, please do so ASAP.
This is a blocker for us.

Once you sort the breaking changes out, you can decide to merge them eventually.

btheosam, 2017-09-08:
Not mine.

btheosam, 2017-09-08:
Heard that the author is aware of this. He will fix soon.

sunghan-chang, 2017-09-08:
Make.defs at os folder was changed to support external protocols like mbedtls, dhcpc, slsi_wifi and so on.
If you didn't update it, let's update it.
Here are methods to do it.
1. Run configure shell again, but it cause initialized .config.
2. Just copy it from build/configs/xxx/yyy folder.

tadziopazur, 2017-09-08:
The issue seems to be a bit deeper than this.
For one, running os/tools/configure.sh does change some git-untracked files; just copying defconfig later is not sufficient.
Second, the menuconfig option is somewhat tricky - not all configurations validated by menuconfig do build. As in: I took defconfig we use for IoT.js, it validated, but failed to build (WIFI_MANAGER and WIFI_NETUTILS were missing).

Other than that, I managed to get things running out of defconfig being very similar to what we used before. This pretty much closes the issue.

closed 2017-09-08 by tadziopazur