Samsung/TizenRT #566: Crash upon boot

tadziopazur, 2017-09-11:
Latest TizenRT (revision 1a6d47ada529) crashes Artik053 board upon boot (log below).
Reverting commit 035144d904ad fixes the problem on my board.

> U-Boot 2017.01-00065-g11a594a84763 (May 04 2017 - 21:26:44 +0900)
> 
> CPU:   Exynos200 @ 320 MHz
> Model: ARTIK-053 based on Exynos T20
> DRAM:  722 KiB
> WARNING: Caches not enabled
> BL1 released at 2017-3-13 15:00
> Flash: 8 MiB
> *** Warning - bad CRC, using default environment
> 
> In:    serial@80180000
> Out:   serial@80180000
> Err:   serial@80180000
> Hit any key to stop autoboot:  0 
> gpio: pin gpg16 (gpio 46) value is 1
> \#\# Starting application at 0x040C8020 ...
> s5j_sflash_init: FLASH Quad Enabled
> uart_register: Registering /dev/console
> uart_register: Registering /dev/ttyS0
> uart_register: Registering /dev/ttyS1
> uart_register: Registering /dev/ttyS2
> uart_register: Registering /dev/ttyS3
> uart_register: Registering /dev/ttyS4
> up_assert: Assertion failed at file:driver/mtd/smart.c line: 1280 task: Idle Task
> up_dumpstate: Current sp: 0206f728
> up_dumpstate: User stack:
> up_dumpstate:   base: 0206f8c4
> up_dumpstate:   size: 00000400
> up_dumpstate:   used: 00000000
> up_dumpstate: User Stack
> up_stackdump: 0206f720: 040caa10 020480ec 00000002 040caa1c 0206f728 020254fc 00000400 0204
> up_stackdump: 0206f740: 042072c4 040caae0 00000500 020255d0 00000500 04228fb0 0000071a 020c
> up_stackdump: 0206f760: 0471a000 041783d0 0206f7a0 00000001 00000004 02074ee0 00000000 0000
> up_stackdump: 0206f780: 00000000 e6074e90 0207cb85 041789f8 0206f7c8 02074ee0 020254fc 0000
> up_stackdump: 0206f7a0: 00000000 03200000 02074f40 04001be8 0207cfa0 04187078 00001000 0000
> up_stackdump: 0206f7c0: 04220000 02000000 fffffffb 00000000 020480ec 0417ba5c 0206f814 0408
> up_stackdump: 0206f7e0: 040d8c04 0206f814 02074e90 00000004 00000004 02074ee0 00000064 0000
> up_stackdump: 0206f800: 0420772c 040cd068 00000000 00000000 01170660 396d6f72 60000100 0200
> up_stackdump: 0206f820: 04001000 00000800 02074e90 00064000 020249cc 042075e3 0420759e 0425
> up_stackdump: 0206f840: 0000077e 00000009 04170f0c 040cd2c8 02072e50 00000000 040d0b1c 0400
> up_stackdump: 0206f860: 04170660 04228c70 04170660 040ccc8c 00000003 040ce1cc 00000003 0418
> up_stackdump: 0206f880: 00000000 020480ec 00000002 040ce24c 00000003 040ce260 02071f80 0408
> up_stackdump: 0206f8a0: 00000024 00078738 0206f8c8 00000020 020480ec 00000000 00000400 0208
> up_stackdump: 0206f8c0: 040c8020 020a7554 00000000 00000000 00000010 80000000 0000dead 0000
> up_registerdump: current_regs is Null, up_assert(panic) was called directly.
> up_registerdump: SPSR_SVC: 60000153      CPSR : 60000153
> up_registerdump: SP_IRQ: 02023800        LR_IRQ: 600001d3        SPSR_IRQ : 60000153
> up_registerdump: SP_ABT: 26305425        LR_ABT: 28804a15        SPSR_ABT : 08015806
> up_registerdump: SP_UND: 0546acc0        LR_UND: 68440882        SPSR_UND : 06030a8a
> up_registerdump: R0: 0202776c 00000020 0000004d 0202776c 020480ec 00000002 040c8020 0000002
> up_registerdump: R8: 020c6884 02044ef0 00000000 00000000 00000000 0206f6f0 040ca64c 040ca54
> up_registerdump: CPSR: 60000153
> up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
> up_taskdump: hpwork: PID=1 Stack Used=284 of 2028
> up_taskdump: lpwork: PID=2 Stack Used=276 of 2028
> up_taskdump: logm: PID=3 Stack Used=324 of 2028
>

Taejun-Kwon, 2017-09-18:
https://github.com/Samsung/TizenRT/pull/593 에 수정된 이슈입니다. smart.c 대신 ftl을 사용하시면 됩니다.

sunghan-chang, 2017-09-20:
That crash was resolved by Taejun's changes. And if you have same crash on new board, clean your user fs partition through below command.
```
cd os
make download ERASE_USERFS
```
I close this. If you have additional question, please reopen it.

closed 2017-09-20 by sunghan-chang