Samsung/TizenRT #570: QEMU Problem.

EunBongSong, 2017-09-12:
I use qemu/tash_16m config.
and then run as below and this stuck as below log.
How can i fix this ?
ebsong@ubuntu:~/tizenrt/github/upstream/TizenRT/os$ qemu-system-arm -M lm3s6965evb -kernel ../build/output/bin/tinyara -nographic -gdb tcp::3333
up_assert: Assertion failed at file:armv7-m/up_blocktask.c line: 111 task: 
up_dumpstate: sp:         20002f38
up_dumpstate: stack base: 00000155
up_dumpstate: stack size: 0000011d
up_dumpstate: ERROR: Stack pointer is not within the allocated stack
mpu_show_regioninfo: *****************************************************************************
mpu_show_regioninfo: *REGION_NO.	BASE_ADDRESS	    SIZE	 REG	   	STATUS	ATTRIBUTES*
mpu_show_regioninfo: *****************************************************************************
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo:        0		       0	       0	       0	       0	       0
mpu_show_regioninfo: *****************************************************************************

manoharahk, 2017-09-12:
1. Please use "qemu-2.10.0" for host, Refer "https://github.com/Samsung/TizenRT/tree/master/build/configs/qemu"

2. If you use "tash_16m" configuration, then you must apply the below patch to "qemu-2.10.0" source.
https://github.com/Samsung/TizenRT/blob/master/build/configs/qemu/qemu-2.10.0-rc2_increase_ram_size.patch

EunBongSong, 2017-09-12:
Thanks for your reply.
It works well after following your guide.

closed 2017-09-12 by EunBongSong