Samsung/TizenRT #763: fstat is not supported

tadziopazur, 2017-10-06:
For one, fstat is mandated by POSIX.1-2001 and later.

There are some leftovers from the fork source (NuttX has and had the fstat at the point of forking) in the form of:
os/include/sys/stat.h:int fstat(int fd, FAR struct stat *buf)
and there is no actual implementation (linking fails):

/opt/iotjs/integration/iotjs.git/deps/tizenrt/os/../build/output/libraries/libtuv.a(fs.c.obj): In function `uv__fs_fstat':
/opt/iotjs/integration/iotjs.git/deps/libtuv/src/unix/fs.c:759: undefined reference to `fstat'
Makefile:187: recipe for target '../build/output/bin/tinyara' failed
make[1]: *** [../build/output/bin/tinyara] Error 1

The same applies to fstatfs (epoll, symlink, link and a few others), although these do not seem to be commonly used.

Taejun-Kwon, 2017-10-10:
I already discussed this issue with iotjs team before, you can use stat.
We will try to apply fstat later.

tadziopazur, 2017-10-10:
I can't use stat, since it works on paths, not open descriptors.
The IoT.js API mandates to have fstat.
Using paths instead of descriptors is hard to do right without OS support.

I posted a back-port from nuttx PR: https://github.com/Samsung/TizenRT/pull/775

sunghan-chang, 2017-11-14:
This was merged so that I close it.

closed 2017-11-14 by sunghan-chang