Samsung/TizenRT #881: UDP broadcast is not supported

tadziopazur, 2017-10-30:
Lets assume there are two sockets
port = 32100;
S1 = socket(INET, DGRAM).bind(0.0.0.0, port);
S2 = socket(INET, DGRAM).bind(0.0.0.0, port);
S2.recv();
S1.sendto(message, {INET, DGRAM, 255.255.255.255, port});

S2 never receives the message.

This is due to:
os/net/lwip/src/core/ipv4/ip.c:167:
struct netif *ip_route(ip_addr_t *dest) {
...
    if (ip_addr_netcmp(dest, &(netif->ip_addr), &(netif->netmask))) {
      return netif;
    }

The function ip_addr_netcmp() does not support broadcast/multicast adresses.

Another problem is that function udp_sendto_chksum (os/net/lwip/src/core/udp.c:474)
calls the ip_route just once. So if there are more network interfaces to broadcast on, lwip implementation of UDP does nota llow that as well.

Should I split this issue into two?

pillip8282, 2017-10-31:
it would be better to split those issues

pillip8282, 2017-11-02:
Did you test broadcast sending at tizenrt? It was working when I tested it.
I implemented a broadcast server sample in linux side and a sender in TizenRT(commit id:9639cc9fc1bb5d98bd67a3bb25b61f6a0e3b1f47).

tadziopazur, 2017-11-06:
We did test the UDP broadcast over loopback and it didn't work.
Namely ip_route() selects a signle network interface to send a packet thru:

> if (ip_addr_netcmp(dest, &(netif->ip_addr), &(netif->netmask))) {
function ip_addr_netcmp() does not care about multicast or broadcast addresses, and the packet is sent nowhere.

For iotjs we have made a work-around:
https://github.com/tadziopazur/TizenRT/pull/1

All in all I think that it'd be useful to integrate recent lwip as sub-module into tizenrt.
This is something that NuttX failed to do, so TizenRT has a chance to do better here.

pillip8282, 2017-11-07:
I didn't test a loopback broadcast. but sending broadcast or multicast to local network succeeds. it would be better if you share your test code.

Furthermore, there are no differences in ip_addr_netcmp() between 2.0.2 and TizenRT version

sunghan-chang, 2018-02-12:
There is no activity. If you need to check this, kindly reopen this with @pillip8282 's request.

closed 2018-02-12 by sunghan-chang