Samsung/TizenRT #884: arastorage can't support update command

ddfb2004, 2017-10-31:
The arastorage's command is like the macros below, but if I want to update the record in the database, how could I do? I haven't found an update command for this.

#define AQL_TYPE_CREATE_ATTRIBUTE
#define AQL_TYPE_CREATE_INDEX    
#define AQL_TYPE_CREATE_RELATION 

#define AQL_TYPE_INSERT          

#define AQL_TYPE_REMOVE_ATTRIBUTE
#define AQL_TYPE_REMOVE_INDEX    
#define AQL_TYPE_REMOVE_RELATION 
#define AQL_TYPE_FLUSH           

#define AQL_TYPE_SELECT          
#define AQL_TYPE_REMOVE_TUPLES

shivgarg, 2017-10-31:
We currently do not support update command. 
As a stopgap arrangement you can use "remove and insert" operation in conjunction to get the equivalent result of an update command

ddfb2004, 2017-10-31:
I tried the command "REMOVE FROM rel WHERE id = 1;" in arastorage_itc module, it cause a crash for this. Is there any problems about the command?

shivgarg, 2017-10-31:
The remove command works fine. Remove was successful in itc_arastorage when i inserted it near line 147 of itc_arastorage_main.c .
`	memset(g_query, 0, QUERY_LENGTH);
	snprintf(g_query, QUERY_LENGTH, "REMOVE FROM %s WHERE id = 1;", RELATION_NAME);
	g_cursor = db_query(g_query);
	if (DB_SUCCESS(cursor_move_first(g_cursor))) {
		do {
			ret = db_print_tuple(g_cursor);
			TC_ASSERT_EQ("db_print_tuple", DB_ERROR(ret), 0);
		} while (DB_SUCCESS(cursor_move_next(g_cursor)));
	}
	TC_SUCCESS_RESULT();
`
. You can check the sample of remove command from this [file](https://github.com/Samsung/TizenRT/blob/master/apps/examples/testcase/ta_tc/arastorage/utc/utc_arastorage_main.c)

ddfb2004, 2017-11-01:
Thanks for your answers, the remove command works correctly.

sunghan-chang, 2017-11-01:
@ddfb2004 Hello. Do you have any additional question? If it isn't, let's close the issue.

closed 2017-11-02 by ddfb2004