Samsung/TizenRT #885: vfork() cant work properly

geek43, 2017-10-31:
Hi.

I'm handling with vfork(), found it can't work properly.
Please focus on below code, After vfork(), OS makes two process, is called parent and child.
After vfork(), parent and child is made successfully. But child is on;ly running, parent can't execute until child will be terminated. If child will be terminated after 10 seconds, then parent process would be executed.

   int pid;

   pid = vfork();

   if ( pid == 0 ) {
       while (1) {
          printf("This is child\n");
          sleep(1);
       }
   } else {
       while (1) {
          printf("This is parent\n");
          sleep(1);
       }
   }

Thanks you.

sunghan-chang, 2017-10-31:
Hello @geek43 .
Vfork calls task_vforkstart and there is waitpid for child inside that. That's why parent can't execute until child is terminated.
This is written in [linux manpage](http://man7.org/linux/man-pages/man2/vfork.2.html) also.
```
vfork() differs from fork(2) in that the calling thread is suspended
       until the child terminates
```

sunghan-chang, 2017-10-31:
@geek43 If you don't have any additional question, please close this.

closed 2017-11-03 by sunghan-chang