Samsung/TizenRT #896: Is the arastorage thread safe?

ddfb2004, 2017-11-02:
I have two questions about the arastorage:
1. Is it thread safe? could I share a database of arastorage between two threads without making any protections of critical section?
2. A relation of arastorage contains only one table, and it couldn't create more tables?

shivgarg, 2017-11-06:
1. No arastorage is not completely thread safe. It is recommended to use locks to prevent concurrency issues.
2. You can create multiple relations in arastorage. Just make sure you have adequate space on file system.

shivgarg, 2017-11-06:
The arastorage limits some things due to the requirement of low memory footprint. To tweak arastorage to use needs you can change various parameters of db in framework/src/arastorage/db_options.h file.

ddfb2004, 2017-11-06:
Thanks for your answers

closed 2017-11-06 by ddfb2004