Samsung/TizenRT #938: Build failure

ravisankarg, 2017-11-10:
[config.txt](https://github.com/Samsung/TizenRT/files/1460811/config.txt)
**build failure**

make[2]: Entering directory '/home/ravi/sbox/gits/github/TizenRT/external/wakaama'
CC:  examples/shared/commandline.c
In file included from examples/shared/commandline.c:18:0:
examples/shared/commandline.c: In function 'prv_displayHelp':
examples/shared/commandline.c:74:17: error: invalid type argument of '->' (have 'int')
         fprintf(stdout, HELP_COMMAND"\t"HELP_DESC"\r\n");
                 ^
examples/shared/commandline.c:78:21: error: invalid type argument of '->' (have 'int')
             fprintf(stdout, "%s\t%s\r\n", commandArray[i].name, commandArray[i].shortDesc);
                     ^
examples/shared/commandline.c:83:17: error: invalid type argument of '->' (have 'int')
         fprintf(stdout, "%s\r\n", cmdP->longDesc?cmdP->longDesc:cmdP->shortDesc);
                 ^
examples/shared/commandline.c: In function 'handle_command':
examples/shared/commandline.c:116:21: error: invalid type argument of '->' (have 'int')
             fprintf(stdout, UNKNOWN_CMD_MSG"\r\n");
                     ^
Makefile:75: recipe for target 'commandline.o' failed
make[2]: *** [commandline.o] Error 1
make[2]: Leaving directory '/home/ravi/sbox/gits/github/TizenRT/external/wakaama'
Makefile:111: recipe for target 'wakaama_all' failed
make[1]: *** [wakaama_all] Error 2
make[1]: Leaving directory '/home/ravi/sbox/gits/github/TizenRT/external'
LibTargets.mk:173: recipe for target '../external/libexternal.a' failed
make: *** [../external/libexternal.a] Error 2

**build options:**
./configure.sh sidk_s5jt200/hello
config file attached!

rzr, 2018-01-23:
This issue is related to missing fd in config file, can you please try to regenerate on current base, if it fails again I'll fix it.

sunghan-chang, 2018-02-26:
Some combination of configs makes compilation and linking errors so that I make a PR #1298 . But that is not enough to resolve all of errors. We will resolve them one by one. Anyway I resolve above erorrs. 
Be aware that there is an error on lwip about tcpip_input. Because lwip will be updated to 2.0.2, we keep it.
If you have any additional question, please make an issue again.
Thank you.

closed 2018-02-26 by sunghan-chang