Samsung/TizenRT #1099: Kernel panic occurred  when using tinyalsa libaray

shijiameng, 2017-12-11:
Hello, I've tired to to play a wave sound by using the tinyalsa library with an ALC5658 codec.  
However, it seems that there are some bugs existing in the driver so that a kernel panic always occurred:

_i2s_send: apb=2064010 nbytes=1024 arg=205f040 timeout=100
i2s_send: apb=2065040 nbytes=1024 arg=205f040 timeout=100
i2s_txp_worker: txp.act.head=0 txp.done.head=205cee0
up_assert: Assertion failed at file:mqueue/mq_msgfree.c line: 146 task: tash
up_dumpstate: Current sp: 02063dc0
up_dumpstate: User stack:
up_dumpstate:   base: 02063fe8
up_dumpstate:   size: 00000fec
up_dumpstate:   used: 0000040c
up_dumpstate: User Stack
up_stackdump: 02063dc0: 02061354 0416f294 0204ddd4 0416f034 02063e04 02061354 02063e10 02063e04
up_stackdump: 02063de0: 80000053 0416ed28 02063e04 0205fe70 00000800 02065040 00000400 0416adbc
up_stackdump: 02063e00: 020621d4 00000001 00000400 02065040 02050001 02064010 00000001 02066070
up_stackdump: 02063e20: 02063e84 0205fe70 00001000 02023d7c 00000003 040e1a48 00000008 02063ef8
up_stackdump: 02063e40: 00000003 00000000 0000001d 040e1af4 00000010 deadbeef 0205fda2 0205fdb6
up_stackdump: 02063e60: 00000000 00000000 00000010 00000002 0000bb80 00000000 00000400 00000002
up_stackdump: 02063e80: 00000010 0205fe70 46464952 025a5ff8 45564157 61746164 025a5f5c 00020001
up_stackdump: 02063ea0: 0000bb80 0002ee00 00100004 020621d4 00000000 02063ef8 00000003 040ded24
up_stackdump: 02063ec0: 040deb34 02024054 02063ef8 040dae08 02063ef8 00000003 00000000 0205fdb9
up_stackdump: 02063ee0: 00000003 00000001 00000000 0000002a 0000002a 040db264 0205fd90 0205fd99
up_stackdump: 02063f00: 0205fda2 00000000 020239f0 040c8194 040ca170 00000000 02025968 02025974
up_stackdump: 02063f20: 00000009 00000000 020239f0 20000053 00000001 00000001 020239f0 00000000
up_stackdump: 02063f40: 00000001 00000000 02063f58 040cdebc 040cc9a0 20000053 020239c4 00000001
up_stackdump: 02063f60: 0205fdba 040ca85c 00000029 00000003 0205fd90 00000008 0000002a 00000003
up_stackdump: 02063f80: 0205fd90 00000008 00000001 040db5f4 02063fb0 deadbeef deadbeef 00000000
up_stackdump: 02063fa0: 00000008 02025f34 deadbeef 00000008 00000006 00000000 deadbeef 00000000
up_stackdump: 02063fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 040cc784
up_stackdump: 02063fe0: 00000000 00000000 deadbeef 02063ff4 00000000 68736174 deadbe00 00000000
up_registerdump: current_regs is Null, up_assert(panic) was called directly.
up_registerdump: SPSR_SVC: 20000053      CPSR : 60000053
up_registerdump: SP_IRQ: 02023800        LR_IRQ: 200000d3        SPSR_IRQ : 20000053
up_registerdump: SP_ABT: 00000000        LR_ABT: fffc0014        SPSR_ABT : 800001d7
up_registerdump: SP_UND: 00000000        LR_UND: 00000040        SPSR_UND : 000001db
up_registerdump: R0: 02026cac 00000000 80180000 02026cac 02026cac 02025850 02026de0 02026d84
up_registerdump: R8: 020614f0 02066070 000002b8 00000001 00000000 02063da0 040c96fc 040c9514
up_registerdump: CPSR: 20000053
up_dumpstate: *******************************************
up_dumpstate: List of all tasks in the system:
up_dumpstate: *******************************************
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=276 of 2028
up_taskdump: lpwork: PID=2 Stack Used=164 of 2028
up_taskdump: LWIP_TCP/IP: PID=3 Stack Used=228 of 4068
up_taskdump: tash: PID=5 Stack Used=1036 of 4076_

How could I solve this problem?

jsdosa, 2017-12-11:
@kishore-sn

shivgarg, 2017-12-12:
Could you give the snippet you are using to play wav file?

shijiameng, 2017-12-12:
I just ported the nxplayer projects from SamsungARTIK/TizenRT.git
You can copy the entire folder into apps/example directory and check it in menuconfig.
Then, execute it as following:

TASH> nxplayer
nxplayer> device /dev/audio/pcmC0D0p   ->Select the playback device
nxplayer> play /mnt/xxx.wav  ->Play the audio file
....kernel panic....

shivgarg, 2017-12-12:
Did you try running the audio utc? Does it work?

shijiameng, 2017-12-12:
I tried the audio utc, it seems ok that I can see the I2S rx process:

_i2s_rx_worker: rx.act.head=0 rx.done.head=205cde8
i2s_receive: apb=2068070 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdcc
i2s_receive: apb=20690a0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdb0
i2s_receive: apb=206a0d0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd94
i2s_receive: apb=206b100 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd78
i2s_receive: apb=206c130 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce3c
i2s_receive: apb=2061590 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce20
i2s_receive: apb=2066010 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce04
i2s_receive: apb=2067040 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cde8
i2s_receive: apb=2068070 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdcc
i2s_receive: apb=20690a0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdb0
i2s_receive: apb=206a0d0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd94
i2s_receive: apb=206b100 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd78
i2s_receive: apb=206c130 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce3c
i2s_receive: apb=2061590 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce20
i2s_receive: apb=2066010 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce04
i2s_receive: apb=2067040 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cde8
i2s_receive: apb=2068070 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdcc
i2s_receive: apb=20690a0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cdb0
i2s_receive: apb=206a0d0 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd94
i2s_receive: apb=206b100 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cd78
i2s_receive: apb=206c130 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce3c
i2s_receive: apb=2061590 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce20
i2s_receive: apb=2066010 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205ce04
i2s_receive: apb=2067040 nmaxbytes=4096 arg=205ec70 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cde8
i2s_receive: apb=2068070 nmaxbytes=4096 arg=205ec70 timeout=100_

shivgarg, 2017-12-12:
We do not plan to support nxplayer application . Many changes have been made to audio framework since we removed nxplayer application from the repository. One of the changes which I found out while going though nxplayer is the change in semantics of buffer size. Since commit bd434bfc95d7542501ba3bc8a3fe9156e4981cf7, the buffer size is reported in number of frames and not in number of bytes. You can use code similar to utc_audio_pcm_writei_p function defined in utc_audio_main.c file to play a file.

shijiameng, 2017-12-12:
I wrote a playback program refer to the utc_audio_pcm_writei_p function, but kernel panic still occurs when try to play a wav file. Could you please to test my code? 

`#include <tinyara/config.h>
#include <tinyara/audio/audio.h>
#include <tinyalsa/tinyalsa.h>
#include <sys/types.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>
#include <errno.h>
#include <assert.h>

#include <fcntl.h>
#include <debug.h>

/****************************************************************************
 * Private Type Declarations
 ****************************************************************************/
typedef struct {
	char	szRiffID[4];		// 'R','I','F','F'
	uint32_t	dwRiffSize;
	char	szRiffFormat[4];	// 'W','A','V','E'
} RIFF_HEADER;

typedef struct {
	uint16_t	wFormatTag;		
	uint16_t	wChannels;
	uint32_t	dwSamplesPerSec;
	uint32_t	dwAvgBytesPerSec;
	uint16_t	wBlockAlign;
	uint16_t	wBitsPerSample;
} WAVE_FORMAT;

typedef struct {
	char		szFmtID[4];		// 'f','m','t',' '
	uint32_t	dwFmtSize;
	WAVE_FORMAT	wavFormat;
} FMT_BLOCK;

typedef struct {
	char		szFactID[4];	// 'f','a','c','t'
	uint32_t	dwFactSize;
} FACT_BLOCK;

typedef struct {
	char 		szDataID[4];			// 'd','a','t','a'
	uint32_t	dwDataSize;
} DATA_BLOCK;

struct audio_session_s {
	FILE *fp;
	struct pcm_config pcm_cfg;
	struct pcm *pcm;
};

static int audio_play(struct audio_session_s *session)
{
	uint8_t *buff;
	size_t buflen, nread;
	int err = OK;
	
	buflen = pcm_frames_to_bytes(session->pcm, pcm_get_buffer_size(session->pcm));
	
	printf("buffer size: %u\n", buflen);
	
	buff = malloc(buflen);
	if (!buff) {
		fprintf(stderr, "%s: failed to allocate memory!\n", __func__);
		err = -ENOMEM;
	}
	
	if (err == OK) {
		do {
			nread = fread(buff, 1, buflen, session->fp);
			if (nread > 0) {
				printf("writing\n");
				if (pcm_writei(session->pcm, buff, pcm_bytes_to_frames(session->pcm, nread)) < 0) {
					fprintf(stderr, "%s: failed to play sample!\n", __func__);
					err = -EIO;
				}
			}
		} while (nread > 0 && !err);
	}
	
	if (buff) free(buff);
	
	return err;
}

static void audio_session_destroy(struct audio_session_s *session)
{
	if (session) {
		pcm_close(session->pcm);
		fclose(session->fp);
		free(session);
	}
}

static struct audio_session_s *audio_session_init(const char *path)
{
	struct audio_session_s *session;
	RIFF_HEADER hdr;
	FMT_BLOCK	fmt;
	DATA_BLOCK data;
	FILE *fp;

	fp = fopen(path, "rb");
	if (!fp) {
		perror(path);
		return NULL;
	}

	fread(&hdr, sizeof(RIFF_HEADER), 1, fp);	
	fread(&fmt, sizeof(FMT_BLOCK), 1, fp);
	
	printf("Channels: %u\n", fmt.wavFormat.wChannels);
	printf("Sample rate: %u\n", fmt.wavFormat.dwSamplesPerSec);
	printf("Average bytes second: %u\n", fmt.wavFormat.dwAvgBytesPerSec);
	printf("Block align: %u\n", fmt.wavFormat.wBlockAlign);
	printf("Bits Per Sample: %u\n", fmt.wavFormat.wBitsPerSample);
	
	char ch = 0;
	
	while (fread(&ch, 1, 1, fp) == 1 && ch != 'd') {}
	fseek(fp, -1, SEEK_CUR);
	
	fread(&data, sizeof(DATA_BLOCK), 1, fp);
	
	printf("Data Size: %u\n", data.dwDataSize);

	session = malloc(sizeof(*session));
	DEBUGASSERT(session);
	
	session->fp = fp;
	session->pcm_cfg.channels = fmt.wavFormat.wChannels;
	session->pcm_cfg.format = PCM_FORMAT_S16_LE;
	session->pcm_cfg.rate = fmt.wavFormat.dwSamplesPerSec;
	session->pcm_cfg.period_size = 1024;
	session->pcm_cfg.period_count = 2;
	
	session->pcm = pcm_open(0, 0, PCM_OUT, &session->pcm_cfg); 
	if (!session->pcm) {
		perror("pcm_open()");
		fclose(fp);
		free(session);
		session = NULL;
	}
	
	if (!pcm_is_ready(session->pcm)) {
		fprintf(stderr, "failed to open PCM device!\n");
		pcm_close(session->pcm);
		fclose(fp);
		free(session);
		session = NULL;
	}
	
	return session;
}

void alc5658_playback_main(int argc, char *argv[])
{
	if (argc < 3)
		return;
		
	struct audio_session_s *session = audio_session_init(argv[2]);
	
	if (session) {
		printf("Playing...\n");
		audio_play(session);
		audio_session_destroy(session);
	}
}
`

shijiameng, 2017-12-12:
Unfortunately, audio_utc is also suffering from the kernel panic...

_[utc_audio_pcm_bytes_to_frames_n] PASS 

[utc_audio_pcm_format_to_bits_p] PASS 

[utc_audio_pcm_format_to_bits_n] PASS 
Sample will be recorded for 3s, press any key to start(Total frame: 48000)

TASH>>i2s_receive: apb=20b1010 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b2040 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b3070 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b40a0 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b50d0 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b6100 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b7130 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_receive: apb=20b8160 nmaxbytes=4096 arg=205ee00 timeout=100
i2s_rx_worker: rx.act.head=0 rx.done.head=205cfcc
up_assert: Assertion failed at file:mqueue/mq_msgfree.c line: 146 task: audio_utc
up_dumpstate: Current sp: 020b0f10
up_dumpstate: User stack:
up_dumpstate:   base: 020b0fe0
up_dumpstate:   size: 00000fe4
up_dumpstate:   used: 000002f4
up_dumpstate: User Stack
up_stackdump: 020b0f00: 80290024 0204dcf4 00000008 020b0f5c 020aca24 04171ad8 0204dcf4 04171878
up_stackdump: 020b0f20: 020b0f5c 020aca24 020b0f68 020b0f5c 80000053 0417156c 020b0f5c 020ac7d0
up_stackdump: 020b0f40: 00001000 00000000 00000400 0416ce68 00000000 041753f8 020ac150 00000001
up_stackdump: 020b0f60: 020a1000 04173294 02050001 020b1010 0202747c 00000005 0202747c 020b9190
up_stackdump: 020b0f80: 020ac7d0 00000005 0000bb80 00000004 00000000 040d8d74 deadbeef deadbeef
up_stackdump: 020b0fa0: deadbeef deadbeef deadbe00 deadbeef deadbeef deadbeef deadbeef 00000000
up_stackdump: 020b0fc0: 00000000 00000000 00000000 00000000 00000000 040cc798 00000000 00000000
up_registerdump: current_regs is Null, up_assert(panic) was called directly.
up_registerdump: SPSR_SVC: 60000053      CPSR : 60000053
up_registerdump: SP_IRQ: 02023800        LR_IRQ: 600000d3        SPSR_IRQ : 60000053
up_registerdump: SP_ABT: 00121000        LR_ABT: 00048284        SPSR_ABT : 800001db
up_registerdump: SP_UND: ffffff7d        LR_UND: 04008224        SPSR_UND : 800001d7
up_registerdump: R0: 02026bcc 00000000 80180000 02026bcc 02026bcc 02025778 02026d00 02026ca4
up_registerdump: R8: 020aae80 00000000 00000000 00000000 00000000 020b0ef0 040c96fc 040c9514
up_registerdump: CPSR: 60000053
up_dumpstate: *******************************************
up_dumpstate: List of all tasks in the system:
up_dumpstate: *******************************************
up_taskdump: Idle Task: PID=0 Stack Used=0 of 0
up_taskdump: hpwork: PID=1 Stack Used=276 of 2028
up_taskdump: lpwork: PID=2 Stack Used=164 of 2028
up_taskdump: LWIP_TCP/IP: PID=3 Stack Used=228 of 4068
up_taskdump: tash: PID=5 Stack Used=684 of 4076
up_taskdump: audio_utc: PID=7 Stack Used=756 of 4068_

sunghan-chang, 2017-12-20:
@shijiameng We have not enough information of yours.
The facts are below:
1. Tizen RT and ARTIK Tizen RT are different.
2. There is no error and no crash on our audio utc.
3. Your base is ARTIK Tizen RT and we don't know what you modify.

We have media player on apps/example. Could you use that instead of nxplayer?

shijiameng, 2018-01-02:
Yes, I fixed a stack overflow bug in the module which I developed and now the panic no longer occurs. But I have another question, that is, how can I un-mute the speaker of alc5658? In alc5658script.h, register ALC5658_SPKOUT doesn't seem to be configured, and may I get the details of alc5658 registers?

btheosam, 2018-01-10:
@shijiameng Please refer alc5658reg.h for alc5658 registers.

btheosam, 2018-01-10:
@shijiameng please refere alc5658reg.h and alc5658scripts.h to know register addresses and values.

sunghan-chang, 2018-01-17:
@shijiameng If you don't have any other question, let's close this.

shijiameng, 2018-01-17:
Ok, I solved the problem. Please close this issue

closed 2018-01-18 by sunghan-chang