Samsung/TizenRT #1128: the "remove from" command can't work well when called for the seconde time in arastorage

ddfb2004, 2017-12-19:
I wrote a sample in itc_arastorage_main.c to test the "remove from" command, the code is here

void itc_arastorage_db_print_tuple_remove(void)
{
    db_result_t ret;
    memset(g_query, 0, QUERY_LENGTH);
    snprintf(g_query, QUERY_LENGTH, "REMOVE FROM %s WHERE id = 1;", RELATION_NAME);
    g_cursor = db_query(g_query);
    if (DB_SUCCESS(cursor_move_first(g_cursor))) {
        do {
            ret = db_print_tuple(g_cursor);
            TC_ASSERT_EQ("db_print_tuple", DB_ERROR(ret), 0);
        } while (DB_SUCCESS(cursor_move_next(g_cursor)));
    }
    TC_SUCCESS_RESULT();

    memset(g_query, 0, QUERY_LENGTH);
    snprintf(g_query, QUERY_LENGTH, "REMOVE FROM %s WHERE id = 2;", RELATION_NAME);
    g_cursor = db_query(g_query);
    if (DB_SUCCESS(cursor_move_first(g_cursor))) {
        do {
            ret = db_print_tuple(g_cursor);
            TC_ASSERT_EQ("db_print_tuple", DB_ERROR(ret), 0);
        } while (DB_SUCCESS(cursor_move_next(g_cursor)));
    }
    TC_SUCCESS_RESULT();
}

The "remove from" can't work well for the second time, the log is here
I think there is something wrong in the "remove from" proccessing functions or maybe my code is not correct?

aql_parse: Parsing "REMOVE FROM rel WHERE id = 1;"
lvm_set_variable: var id = 0
lvm_print_code: Code: print_operator: = print_operand: var(id):0 print_operand: long:1
db_query: DB : flush insert buffer!!
relation_load: DB: Rel rel, Cardinality 10
aql_init_handle: init handle!
relation_select: relation_select... optype = 131082
storage_get_relation: DB: relation : db-rem
storage_get_relation: DB: relation : db-rem
storage_drop_relation: Unlink rel = db-rem, tuple =
storage_drop_relation: Failed to remove relation file : db-rem
storage_put_relation: DB: Saved relation db-rem tuplefile = tup.514c
relation_load: DB: Rel db-rem, Cardinality 0
index_load: DB: Attempting to load an index over rel.id
relation_select: DB: Found attribute id in relation rel
storage_put_attribute: DB: put_attribute(db-rem, id)
index_load: DB: Attempting to load an index over rel.date
index_load: DB: Failed load an index descriptor from storage
relation_select: DB: Found attribute date in relation rel
storage_put_attribute: DB: put_attribute(db-rem, date)
index_load: DB: Attempting to load an index over rel.fruit
index_load: DB: Failed load an index descriptor from storage
relation_select: DB: Found attribute fruit in relation rel
storage_put_attribute: DB: put_attribute(db-rem, fruit)
derive_relation: variable id 0, value 1
derive_relation: derivation max = 1, min = 1
select_index: DB: The search range for attribute "id" comprises 1 values
index_get_iterator: DB : Index_get_iterator min = 1, max = 1
index_get_iterator: DB: Acquired an index iterator for rel.id over the range (1,1)
storage_get_row: read row = , r = 38
storage_get_row: DB: Read 38 bytes from relation rel
eval_logic: Result1: 1
Result2: 1
lvm_execute: The statement is true
storage_get_row: read row = , r = 38
storage_get_row: DB: Read 38 bytes from relation rel
eval_logic: Result1: 2
Result2: 1
lvm_execute: The statement is false
[... 165 lines cut ...]
eval_logic: Result1: 8
Result2: 2
lvm_execute: The statement is false
storage_get_row: read row = , r = 38
storage_get_row: DB: Read 38 bytes from relation rel
eval_logic: Result1: 9
Result2: 2
lvm_execute: The statement is false
storage_get_row: read row = , r = 38
storage_get_row: DB: Read 38 bytes from relation rel
eval_logic: Result1: 10
Result2: 2
lvm_execute: The statement is false
storage_get_row: DB : tuple_id : 33989764 nrows : 33981808
relation_process_remove: DB: Finished removing tuples. Result relation has 8 tuples
storage_flush_insert_buffer: DB : Stored buffer(304 bytes) in storage : 10
relation_process_remove: DB : flush insert buffer!!
relation_load: DB: Rel rel, Cardinality 9
relation_process_remove: DB: Failed to rename newly created relation
relation_process_result: DB: Failed to process tuples : -3
db_query: DB: Failed to process cursor tuples
storage_unload: DB: Unload tuple file tup.514c
storage_unload: DB: Unload tuple file tup.7c97
aql_deinit_handle: deinit handle!
cursor_move_to: Empty Cursor

shivgarg, 2017-12-19:
I looked into itc. The itc does have this function "itc_arastorage_db_print_tuple_remove".
I think you are using an older version of it. But I tested  consecutive remove statements in arastorage in utc_arastorage_main.c. I inserted the following snippet at line 425 and it worked without any error.
`        /* Remove operation */
         snprintf(query, QUERY_LENGTH, "REMOVE FROM %s WHERE date > 2000 AND date < 8000;", RELATION_NAME2);
         check_query_result(query);
 
         snprintf(query, QUERY_LENGTH, "REMOVE FROM %s WHERE date < 2000 AND date > 1500;", RELATION_NAME2);
         check_query_result(query);
 
         snprintf(query, QUERY_LENGTH, "REMOVE FROM %s WHERE date < 1500 AND date > 1000;", RELATION_NAME2);
         check_query_result(query);
 
         snprintf(query, QUERY_LENGTH, "REMOVE FROM %s WHERE date < 1000;", RELATION_NAME2);
         check_query_result(query);

`

According to the logs, the tc failed at removing the old relation file to create a new file with the remaining tuples. Could you check that you have all the latest changes of arastorage?

ddfb2004, 2017-12-20:
yes，that‘s right. I check out the latest version, the issue has gone. Thanks for your answer.

closed 2017-12-20 by ddfb2004