Samsung/TizenRT #1189: wifi app connection fails with 201 when providing ssid & passphrase in TASH

pholat, 2018-01-12:
Trying to connect WiFI with tash `wifi` command with following setup:
`wifi startsta` -> reply success
`wifi connect MyNetwork MyPassphrase wpa2_aes` -> fails with error -201

Where 201 error in slsi_wifi_api.h is `SLSI_REASON_NETWORK_CONFIGURATION_NOT_FOUND` , whereas my understanding is it was provided in `wifi connect` command.

pholat, 2018-01-15:
My bad, I've been providing data with typo. 

Though logs are a bit missleading in tash command : before connection there were ssid/passwd shown and when it failed there was written that empty ssid/passwd was provided, log:
`Connected to network: bssid: , ssid:
Join failed - reason 201`

dr-venkman, 2018-01-15:
Could you please try as follows:

wifi join Network password wpa2_aes

sunghan-chang, 2018-01-18:
@pholat Is it working with @dr-venkman 's suggestion?

pholat, 2018-02-19:
@sunghan-chang My bad, it's working now - just log was misleading a bit. There was bad character copied into password and that's the error output.

closed 2018-02-19 by pholat