Samsung/TizenRT #1283: No binary for Mac OS artik05x_AppCodesigner

pholat, 2018-02-19:
CodeSigner tool can't be run on MacOS, so I can't work with Artik053s on Mac ( or Windows ), which in other parts of scripts is supported.

sunghan-chang, 2018-02-20:
@pholat I am so sorry but that depends on chip vendor. Let's wait his response.

JunhwanPark, 2018-02-20:
Hi @pholat ,
I prepared it. But I have not updated it yet. It will be updated before February is over. And then I'll tell you again.

pholat, 2018-02-20:
Thanks, for now I'll try to deal with VB machine.

sunghan-chang, 2018-02-21:
@pholat Could I leave a query to you? Do you use the TizenRT on MAC? Could you let me know any problem or difficulty?

pholat, 2018-02-21:
@sunghan-chang Sure :) I've to work on Mac, there was some fun involved for me to i.e. use kconfig on mac, though nothing too serious. I can make PR to README with update how to install it.

While I couldn't configure ftdi driver on mac to have both jtag and uart available, I just use com on hardware uart0 instead - I believe it's common macbook problem. 

I've tried a few examples with ease.
In code documentation in form of readme.md is getting better and actually being helpful 👍 

To conclude my problems usually aren't macbook related :) 

I had two general concerns till now:
*  combination of bash scripts called from Make triggers a bit too many compilations and have some issues combined with ccache. 
*  it's possible to use c++ compiler, and that is great because it has more checks available etc. ( I actually depend on these ), I find adding any standard library troublesome due to headers conflicts.

All in all I didn't have much trouble and rather enjoy the platform.

juitem, 2018-02-21:
It would be appreciated if you could contribute PR for MAC.
Actually we were also considering to support MAC officially on this year, if you proceed with submitting PRs and any additional supports we'll develop.
Thanks

sunghan-chang, 2018-03-06:
@JunhwanPark Could you update supporting codeSigner for mac again? There is no relevant commit in SamsungARTIK.

JunhwanPark, 2018-03-06:
@sunghan-chang Sorry. Update is delayed due to various issues. Please wait a little longer. The goal is to release a new version in mid-March.

JunhwanPark, 2018-03-12:
@sunghan-chang https://github.com/SamsungARTIK/TizenRT/commit/de1e0c1aa02c7a0a3080a72a8c0001a8606855b1

JunhwanPark, 2018-03-12:
Hi @pholat ,
I updated it. Enjoy~

rzr, 2018-03-12:
For the record I think it can be download from SDK archive, is it for linux at least.

I can give more details if needed

sunghan-chang, 2018-03-15:
@JunhwanPark Thank you for supporting.
@pholat I cherry-picked it from SamsungARTIK, ours is https://github.com/Samsung/TizenRT/commit/f61d3ff73aee0d72c96c9a14ef7a371020905ce0. If you configure MACOS as a host, codesigner for mac will be used automatically by download script.

sunghan-chang, 2018-03-15:
@pholat If it is enough to this issue, let's close this.

sunghan-chang, 2018-03-15:
@pholat Due to my fault (wrong modification on f61d3ff), I merged one more commit. https://github.com/Samsung/TizenRT/commit/5d8b5ab094ca1b5bcd8eed2dec21f827aaf29903. Sorry for inconvience.

pholat, 2018-03-16:
@sunghan-chang No worries, thanks for fixing. I actually first moved onto Linux VM and now on linux PC entirely.

sunghan-chang, 2018-03-20:
@pholat Ok, then I close this. If you have any issue on this, please reopen. Thank you for confirmation.

closed 2018-03-20 by sunghan-chang