Samsung/TizenRT #1328: gpio numbering doesn't reflect documentaion on artik053

pholat, 2018-03-14:
When I check errno on opening, setting fcntl, writting to GPIO, I get errors on it, even though it works ( this is: toggles pins ) just right.

Plus:
Using posix `open -> write -> fcnt -> close` api, you need to specify pin number according to:
`arch/arm/src/artik05x/src/artik05x_boot.c` documentation `HW_Datasheet_ARTIK_053s_V1.4.pdf` at section Module pad definitions have it wrongly numbered - i.e. XGPIO13 is there GPIO45 where you have to actually write ‘1’ to /dev/gpio42 to write light it up.

IOTBUS shall be affected just the same.

sunghan-chang, 2018-03-15:
@pholat For first question,
I tested what you mentioned in linux.
```
int test2(void)
{
	return 0;
}
int main(void)
{
	int ret;
	printf("test init errno = %d\n", errno);
	ret = close(100);
	printf("test second ret = %d, errno = %d\n", ret, errno);
	ret = test2();
	printf("test third ret = %d, errno = %d\n", ret, errno);
	return 0;
}
```
It shows as shown below:
```
test init errno = 0
test second ret = -1, errno = 9
test third ret = 0, errno = 9
```
So, errno keeps previous value until updating by another error.
When you try to see errno value with function success, showing some value on errno is normal.
It is same as ours.

sunghan-chang, 2018-03-15:
@JunhwanPark Could you check @pholat 's second query for GPIO?
By his comment, it looks mismatched.

JunhwanPark, 2018-03-15:
@pholat Oh! I see...  I will talk to the person in charge of the document. Thx for reporting.
u can check https://github.com/SamsungARTIK/TizenRT#gpio-mapping .

pholat, 2018-03-15:
@JunhwanPark  Great thanks for confirmation! It will be great to fix docs as these are on the Samsung main page on Artik053 - it's one of first solid info one get on the Artik053 platform.

@sunghan-chang  I've cleaned errno prior to calling open, write etc. just in case, if it wouldn't be too much could you please check with Artik053s too? ( secure one?) 
Also I've freezed updating TizenRT and am on version: 34d5b12f4c3b506e627ef80e12f0a7fbc6f33f40 ( 23 Feb ) - as I need to get code working on our devices prototypes. Maybe there was some fixes on that meantime, sorry for inconvenience :) 

Errors I get on strerror are:
> open '/dev/gpio53' -> no such file or directory
> ioctl -> unknown error
> write -> operation not permitted 
> close -> 69087312
Even thought it works - in which I mean blinks led.

pholat, 2018-03-15:
btw: I'm impressed - documentation is just getting better 👍

sunghan-chang, 2018-03-15:
@pholat Let's clearfy your problem.
1.  They, open, read, write and etc, return ERROR (or negative value).
2. They return OK but when you check errno, it has error value.
Could you let me know what is your problem?

pholat, 2018-03-15:
@sunghan-chang The first one.

sunghan-chang, 2018-03-16:
@pholat Thank you for confirmation. I will check it but my computer was gone. I may do it next week. After checking, I will let you know.

pholat, 2018-03-16:
@sunghan-chang I'm sorry for your loss :) Thanks! I'll let you know if I find something more.

sunghan-chang, 2018-03-21:
@pholat Hello. Sorry for delay. I restore my PC yesterday.
I checked iotbus_test sample which is in apps/examples. There is no issue and it works well without error return. As you know, actually iotbus gpio also uses open, read, write and etc inside.
To know what problem is, could you share your sample code?

pholat, 2018-03-21:
Sure, I use some variant  of gpio_write found in repository with tiny modifications.
```
void write_gpio(const int N, bool state)
 {
     static char dev[16] = {0};
     static char nn[4]   = {0};
     sprintf(nn, "%d", state);
     sprintf(dev, "/dev/gpio%d", N);
     int fd, ret;
     errno = 0;
     fd    = open(dev, O_RDWR);
     if (fd < -1) 
            printf("open fail %d %s %s\n", fd, strerror(errno), dev);
     ret = ioctl(fd, GPIOIOC_SET_DIRECTION, GPIO_DIRECTION_OUT);
     if (ret < -1)
           printf("ioctl err %d: %s\n", ret, strerror(errno));
     ret=write(fd, nn, strlen(nn)));
     if ( ret < -1)
           printf("write err! %d %s\n", ret, strerror(errno));
     ret=close(fd);
     if (ret <-1)
          printf("close fail...%d %s\n", ret, strerror(errno));
}
```

then logs are:
> open fail 6 Unknown error /dev/gpio53
> write err! 1 Unknown error
> open fail 6 Unknown error /dev/gpio53
> write err! 1 Unknown error

Previous logs were messed up due to my own stupidity, sorry about that, I've put wrong printf specifiers.

pholat, 2018-03-22:
This is awkard, I use gcc in version:
> arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 
> (release) [ARM/embedded-7-branch revision 255204]
> Copyright (C) 2017 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

And if(...) without {} then it doesn't work. Please see that if was for negative numbers and positive one is logged. When I've added {} around logs its not happening. 
I'm compiling on Ubuntu: 4.4.0-116-generic #140-Ubuntu SMP Mon Feb 12 21:23:04 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

So with docs fixed it can be closed I think, sorry for confusion.

sunghan-chang, 2018-03-23:
@pholat Thank you for confirmation.

closed 2018-03-23 by sunghan-chang