Samsung/TizenRT #1337: SCSC watchdog can't be disabled in config - this breaks debugging in gdb

pholat, 2018-03-16:
https://github.com/Samsung/TizenRT/blob/ef26351d38194f571f86b981e5e2c9e6efabccfd/os/drivers/wireless/scsc/misc/platform_mif.c#L201 There is no guard to disable this watchdog. This way when debugging, after wifi enable it kicks in and ruins debugging. 

Also I've an issue with stepping on artik053s in generall. I start debug like that:
-> set V=2 in Make.defs
-> set CONFIG_DEBUG_SYMBOLS in .config ( it's used in Make.defs to check if -g flag is used for compilation and linking ) 
-> make & download
-> start gdb server with: `cd TizenRT/build/configs/artik05x/scripts/ && tools/openocd/linux64/openocd -f artik05x.cfg -d`
-> start gdb session with arguments:
`target remote localhost:3333
file TizenRT/build/output/bin/tinyara
monitor reset halt
b whatever_breakpoint_you_choose
c`

Then I cant: 
* step through program - gdb just looses itself. b 
* set breakpoint on up_stackdump() - as I would love to see crashdump unvinded in gdb
I would be grateful for some info how could I achieve that :)

sunghan-chang, 2018-03-21:
@pholat Sorry for delay. I will check and come back to you with result.

sunghan-chang, 2018-03-28:
@pholat Hello. I am sorry for delay. But we had some urgent issue inside (now resolved).
1. Chip vendor, ARTIK, provided scsc module and they want to maintain it. So that it is difficult to modify scsc to support scsc watchdog. In this case, you had better ask it on SamsungARTIK.
2. I've tested GDB with assert.
Codes are
```
int hello_main(int argc, char *argv[])
{
	sleep(20);
	ASSERT(0);
}
```
I used GDB command as shown below:
```
@ubuntu:~$ arm-none-eabi-gdb TizenRT/build/output/bin/tinyara
GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.0.20150604-cvs
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from TizenRT/build/output/bin/tinyara...done.
(gdb) target remote:3333
Remote debugging using :3333
0x00003834 in ?? ()
(gdb) monitor reset halt
target halted in ARM state due to debug-request, current mode: Supervisor
cpsr: 0x60000153 pc: 0x040db8a8
D-Cache: enabled, I-Cache: enabled
(gdb) monitor cortex_r4 maskisr on
cortex_a interrupt mask on
(gdb) b hello_main
Breakpoint 1 at 0x40d30fc: file hello_main.c, line 196.
(gdb) b up_assert
Breakpoint 2 at 0x40c9364: file armv7-r/arm_assert.c, line 945.
(gdb) c
Continuing.

Breakpoint 1, hello_main (argc=1, argv=0x202fd1c) at hello_main.c:196
196	{
(gdb) s
197		sleep(20);
(gdb) s
sleep (seconds=seconds@entry=20) at unistd/lib_sleep.c:155
155		if (seconds) {
(gdb) c
Continuing.

Breakpoint 2, up_assert (filename=filename@entry=0x40ea757 "hello_main.c", 
    lineno=lineno@entry=198) at armv7-r/arm_assert.c:945
945		lldbg("Assertion failed at file:%s line: %d task: %s\n", filename, lineno, this_task()->name);
(gdb) 
```

sunghan-chang, 2018-03-28:
@pholat Is this enough?

closed 2019-04-03 by sunghan-chang