Samsung/TizenRT #1417: media_record doesn't work

shijiameng, 2018-04-03:
I'm testing the media_record library with an ALC5658 codec, the sampled PCM data are always zeros.
Please check it.

jsdosa, 2018-04-03:
how to test?
Please explain to me in detail.

shijiameng, 2018-04-04:
I'm not known if you have an alc5658 codec and test the driver before its publication . In our situation, we connected the alc5658 codec to ARTIK053's I2S interface,  then tested the media_play and media_record through media_test application. Now, the media_play application works well but media_record doesn't work.

Taejun-Kwon, 2018-04-04:
Currently we write data on flash for every record, but performance of artik053 is little slow.
I think you'd better use ramfs instead of smartfs for current record.
We will update latest media f/w and sample application soon.

closed 2018-06-14 by shijiameng