Samsung/TizenRT #1500: libcxx cannot be built

shijiameng, 2018-04-23:
Hello,

I have updated the ARM toolchain to  version 6 (gcc-arm-none-eabi-6_2-2016q4) and trying to build the libcxx library, but there are some errors:

./any.cxx:27:15: fatal error: any: No such file or directory
 #include "any"

/opt/gcc-arm-none-eabi-6+2-2016q4/arm-none-eabi/include/sys/_types.h:168:5: error: 'wint_t' does not name a type wint_t __wch;

How could I tackle these errors?

sunghan-chang, 2018-04-23:
@shijiameng Could you share your Make.defs which is in build/configs/xxx/yyy folder?

shijiameng, 2018-04-23:
I am using the configuration of build/configs/artik053/audio

sunghan-chang, 2018-04-24:
@shijiameng Current artik053/audio does not have C++ config enabled. If you want to use c++ based on artik053/audio, you should turn it on. But we recommand using artik055s/audio. Actually artik053 EVK board does not support audio (artik053 module supports).

jsdosa, 2018-04-24:
@shijiameng  
please check : https://github.com/Samsung/TizenRT/pull/1533/
we do not support the artik053/audio

sunghan-chang, 2018-05-04:
Is there any update? If you have, please reopen this. Thank you.

closed 2018-05-04 by sunghan-chang