Samsung/TizenRT #1637: some configurations build not done

sksno1, 2018-05-04:
Hi 
I have ARTIK053 board. and I test  to build TizenRT offered options(TizenRT/os/tools/configure.sh ).
Some build options have not effect. there is a list

build complete

artik053/minimal 
artik053/hello
artik053/tc
artik053/nettest
artik053/kernel_sample
artik053/iotjs
artik053/helloxx
artik053/cxxtest

build error

artik053/st_things

AR:   net_sockets.o net_close.o net_dupsd.o net_dupsd2.o net_clone.o net_vfcntl.o bsd_socket_api.o recvmsg.o sendmsg.o net_checksd.o netdev_ioctl.o netdev_findbyname.o netdev_count.o netdev_foreach.o netdev_sem.o ethernetif.o net_initialize.o net_lock.o def.o init.o mem.o memp.o netif.o ip.o dns.o timeouts.o pbuf.o raw.o stats.o sys.o tcp.o tcp_in.o tcp_out.o udp.o inet_chksum.o api_lib.o api_msg.o err.o netbuf.o netifapi.o sockets.o tcpip.o netdb.o ethernet.o slipif.o snmp_asn1.o snmp_core.o snmp_mib2.o snmp_mib2_icmp.o snmp_mib2_interfaces.o snmp_mib2_ip.o snmp_mib2_snmp.o snmp_mib2_system.o snmp_mib2_tcp.o snmp_mib2_udp.o snmp_msg.o snmp_netconn.o snmp_pbuf_stream.o snmp_raw.o snmp_scalar.o snmp_table.o snmp_threadsync.o snmp_traps.o autoip.o etharp.o icmp.o igmp.o ip4.o ip4_addr.o ip4_frag.o dhcp.o dhcps.o auth.o chap.o chpms.o fsm.o ipcp.o lcp.o magic.o lwip_md5.o pap.o ppp.o ppp_oe.o randm.o vj.o sys_arch.o
make[1]: Leaving directory '/home/cyclone5/Documents/TizenRT/TizenRT/os/net'
make: *** No rule to make target '../external/iotivity/iotivity_1.2-rel/out/tizenrt/armv7-r/release/liboctbstack.a', needed by '../build/output/libraries/liboctbstack.a'.  Stop.

artik053/extra

lwip/src/netif/slipif.c: In function 'slipif_init':
lwip/src/netif/slipif.c:349:18: error: 'NETIF_FLAG_POINTTOPOINT' undeclared (first use in this function)
  netif->flags |= NETIF_FLAG_POINTTOPOINT;
                  ^~~~~~~~~~~~~~~~~~~~~~~
lwip/src/netif/slipif.c:349:18: note: each undeclared identifier is reported only once for each function it appears in
Makefile:122: recipe for target 'slipif.o' failed
make[1]: *** [slipif.o] Error 1
make[1]: Leaving directory '/home/cyclone5/Documents/TizenRT/TizenRT/os/net'
LibTargets.mk:85: recipe for target 'net/libnet.a' failed
make: *** [net/libnet.a] Error 2

artik053/iotivity

AR:   net_sockets.o net_close.o net_dupsd.o net_dupsd2.o net_clone.o net_vfcntl.o bsd_socket_api.o recvmsg.o sendmsg.o net_checksd.o netdev_ioctl.o netdev_findbyname.o netdev_count.o netdev_foreach.o netdev_sem.o ethernetif.o net_initialize.o net_lock.o def.o init.o mem.o memp.o netif.o ip.o dns.o timeouts.o pbuf.o raw.o stats.o sys.o tcp.o tcp_in.o tcp_out.o udp.o inet_chksum.o api_lib.o api_msg.o err.o netbuf.o netifapi.o sockets.o tcpip.o netdb.o ethernet.o slipif.o snmp_asn1.o snmp_core.o snmp_mib2.o snmp_mib2_icmp.o snmp_mib2_interfaces.o snmp_mib2_ip.o snmp_mib2_snmp.o snmp_mib2_system.o snmp_mib2_tcp.o snmp_mib2_udp.o snmp_msg.o snmp_netconn.o snmp_pbuf_stream.o snmp_raw.o snmp_scalar.o snmp_table.o snmp_threadsync.o snmp_traps.o autoip.o etharp.o icmp.o igmp.o ip4.o ip4_addr.o ip4_frag.o dhcp.o dhcps.o auth.o chap.o chpms.o fsm.o ipcp.o lcp.o magic.o lwip_md5.o pap.o ppp.o ppp_oe.o randm.o vj.o sys_arch.o
make[1]: Leaving directory '/home/cyclone5/Documents/TizenRT/TizenRT/os/net'
make: *** No rule to make target '../external/iotivity/iotivity_1.2-rel/out/tizenrt/armv7-r/release/liboctbstack.a', needed by '../build/output/libraries/liboctbstack.a'.  Stop.

artik053/grpc

CXX:  helloworld.pb.cc
protoc -I . --grpc_out=. --plugin=protoc-gen-grpc=`which grpc_cpp_plugin` helloworld.proto
/bin/sh: 1: protoc: not found
Makefile:89: recipe for target 'helloworld.grpc.pb.cc' failed
make[2]: *** [helloworld.grpc.pb.cc] Error 127
make[2]: Leaving directory '/home/cyclone5/Documents/TizenRT/TizenRT/apps/examples/grpc_greeter_client'
Makefile:109: recipe for target 'examples/grpc_greeter_client_all' failed
make[1]: *** [examples/grpc_greeter_client_all] Error 2
make[1]: Leaving directory '/home/cyclone5/Documents/TizenRT/TizenRT/apps'
LibTargets.mk:158: recipe for target '../apps/libapps.a' failed
make: *** [../apps/libapps.a] Error 2

What I do? please let me know how to build it.

Regards. Shin.

sunghan-chang, 2018-05-04:
Hello @sksno1 .
First of all, sorry for unconvenience.
1. artik053/st_things
2. artik053/iotivity
    They have iotivity enabled. To build iotivity, scons is mandatory tool.
    Could you check [README of iotivity](https://github.com/Samsung/TizenRT/tree/master/external/iotivity#prerequisites)?

3. artik053/grpc
    grpc includes protobuffer and ```protoc``` is needed.
    Find [README](https://github.com/Samsung/TizenRT/blob/master/external/protobuf/README.md).

4. artik053/extra
    1. errors on network
        This is our fault. Network was updated but extra isn't. 
        Temporarilly, could you enter ```make menuconfig``` before executing ```make```?
        That will update .config which shows defined configuration set and build will be done.
    2. errors on iotjs
        Find [prerequisite of iotjs](https://github.com/Samsung/TizenRT/blob/master/external/iotjs/docs/build/Build-for-STM32F4-NuttX.md#1-prepare-for-prerequisite)

sunghan-chang, 2018-05-08:
@sksno1 FYI. I raise a PR #1657  to show necessary package to build or download.
i.e.
```
TizenRT/os/tools$ ./configure.sh artik053/grpc
  Copy build environment files
 
  CAUTION!! To build artik053/grpc, protoc should be installed
  Find details at https://github.com/Samsung/TizenRT/blob/master/external/protobuf/README.md
 
  Configuartion is Done!
```

sunghan-chang, 2018-05-08:
@sksno1 If you don't have any additional question, please reopen this.

closed 2018-05-08 by sunghan-chang