Samsung/TizenRT #1690: bchardev freezes on close on OTA partiton

pholat, 2018-05-10:
I need OTA functionality for our application and added some (dirty WIP) code for just that, as well as updated example in apps/examples directory here:
   https://github.com/pholat/TizenRT/tree/ota_tests 

I've just one issue in line:
    `os/drivers/fota/fota_driver.c:192      close(fota_fd);`
This close closes bchrdev created on fota partition driver, though for some reason in hangs. 

When I restart the board ota partion is being read and app is being updated. I'm quite convinced that it shouldn't be hanging itself on close I'll be very glad to any pointers how it could be possibly fixed, so I could clean it up and make PR with OTA working fine.

jeongchanKim, 2018-05-16:
I checked your code(https://github.com/pholat/TizenRT/tree/ota_tests), but that is not from our repo.
I'm sorry but we cannot cover not from ours.

pholat, 2018-05-16:
@jeongchanKim I don't understand what you need not from your repo? It's forked form TizenRT here. II've just added lacking driver as I need it, though have some issue with bchdev which can't seem to get closed.
Or do you mean it's part of legit Nuttx code TizenRT developers do not maintain?

jeongchanKim, 2018-05-21:
In our code, there is no issue for fota without  bchdev. So I don't know why you added the bch.

pholat, 2018-05-21:
@jeongchanKim there was no fota driver at all, that's why I've used bchdev to write the update onto the flash. There was no issue, because there was no fota driver at all. If it was there - I would have never tried to do it myself.

jeongchanKim, 2018-05-23:
Okay I understand what you're saying. 
But, which board do you use?
If you use artik, fota driver is not required for current version. Because artik05x_configure_partitions automatically set the driver.
If you don't use artik, you are right, we should check fota driver code.

pholat, 2018-07-24:
@jeongchanKim 
Sorry for late response - I missed the reply :) 
I use Artik053s - I needed something to write OTA given binary to OTA partition, so I've added simple code using bchdev to write the data onto this partition. Then when I call flush on bchdev on the mentioned partition at the end of writes it just hangs.

I hope I was 100% clear now :)