Samsung/TizenRT #1865: C++ problems

shijiameng, 2018-06-14:
Hi, 
I'm developing an application by C++ and selected the 'Have C++ compiler' option in the menuconfig. But it cannot be complied with the following errors:

make[1]: Entering directory '/home/sjm/Projects/ARTIK05x/TizenRT/external/libcxx'
./any.cxx:27:15: fatal error: any: No such file or directory
 #include "any"

make[1]: Entering directory '/home/sjm/Projects/ARTIK05x/TizenRT/lib/libxx'
make[1]: *** No rule to make target 'libxx_delete_sized.cxx', needed by 'libxx_delete_sized.o'.  Stop.
make[1]: Leaving directory '/home/sjm/Projects/ARTIK05x/TizenRT/lib/libxx'
LibTargets.mk:149: recipe for target 'libxx/libcxx.a' failed
make: *** [libxx/libcxx.a] Error 2

The toolchain is gcc-arm-none-eabi-6-2017-q1-update.

How can I fix these errors?

manoharahk, 2018-06-14:
C++ requires to enable multiple configs and modification in each Make.defs as well.
Please refer artik053/cxxtest, defconfig & Make.defs

manoharahk, 2018-09-23:
Hello @shijiameng ,

I hope you found the solution. 
Since there is no activity on this issue, I will be closing the issue. Please re-open if you met any problem.

closed 2018-09-23 by manoharahk