Samsung/TizenRT #1903: strncat() without bounds

RootUp, 2018-07-03:
Bug Type: Security
File: https://github.com/Samsung/TizenRT/blob/master/apps/examples/testcase/le_tc/kernel/tc_libc_string.c#L473

i.e

`res_ptr = strncat(dest_arr, src, BUFF_SIZE);` 

 Easily used incorrectly (e.g., incorrectly computing the correct maximum size to add).

Consider strcat_s, strlcat, snprintf, or automatically resizing strings. I feel the risk is high because the length parameter appears to be a constant, instead of computing the number of characters left.

Request team to please have a look and validate.

Regards
Dhiraj

sunghan-chang, 2018-07-09:
@RootUp Thank you for your advise.
We will check almost strncat and similar stubs. But tc is designed to check API operation itself. Because we don't care extension of it, we don't want to calculate.

sunghan-chang, 2018-07-11:
@RootUp We found an issue on http and resolve it #1920 . Thank you again. If you have any query or advice for this, please kindly reopen this.

closed 2018-07-11 by sunghan-chang