Samsung/TizenRT #2163: kernel/mqueue: g_desfree is not initialized!?

zhouxinhe, 2018-09-05:
should we add `sq_init(&g_desfree);` in `void mq_initialize(void)`?

jeongchanKim, 2018-09-05:
g_desfree is a global variable. So it will be initialized automatically.

sunghan-chang, 2018-09-11:
@zhouxinhe Thank you for checking it.
@jeongchanKim Please confirm whether mqueue is not necessary to re-initialize or not.

sunghan-chang, 2018-09-11:
Me and Jeongchan have checked it. There is no difference between adding sq_init and keeping now. But it looks strange because other lists are initialized with sq_init at mq_initiailize function. Let's keep current code.
@zhouxinhe If you have different opinion, please let us know.

zhouxinhe, 2018-09-11:
I faced a bss region corruption issue days ago, crash happened when traversing this queue `g_desfree`...
I checked and found it's not initialized with sq_init (so I opened this issue). 
Now I have found the root cause, which is not related to g_desfree initialization.
I will close this issue, thanks for checking.

closed 2018-09-11 by zhouxinhe