Samsung/TizenRT #2170: [Artik053s] CONFIG_SCSC_WLAN_AUTO_RECOVERY not working - platform infinite hang

pholat, 2018-09-06:
There are times when WiFi hangs, i.e. I saw it happen after non clean reboot ( i.e. after pressing reset ) about 10% of time. 

This results in log:
"HIP state set to #SLSI_HIP_STATE_BLOCKED#
from:
os/TizenRT/os/drivers/wireless/scsc/cm_if.c function: static void slsi_hip_block_bh(struct slsi_dev *sdev) 

after that board freezes and only manual restart helps. When I added plain reboot with:
`boardctl(BOARDIOC_RESET, EXIT_SUCCESS)`

prior to that log , it hangs there. I will mention here that EXIT_SUCCESS is dead parameter because underneath:
`static void up_systemreset(void)` 
is being used, which puts some register and makes infinite hang. My best guess then is that it's really infinite.

I'll add that normally `static void up_systemreset(void)` works fine and it hangs only when WiFi fails.

sunghan-chang, 2018-09-11:
@JunhwanPark Could you check this issue for ARTIK wifi driver?

JunhwanPark, 2018-09-11:
Hi @pholat ,

I have to check the whole log and I will be able to guess. And this repository has a very old version of the `wlanfw.bin` file. It has been updated a few times, so please check it. https://github.com/SamsungARTIK/TizenRT/commit/a6853b4271f9a4a8ba521bda42d5ce0bb56c75f2

It may not be solved by simply changing the WiFi firmware. Please check first in the fork version of this repository. (https://github.com/SamsungARTIK/TizenRT)

pholat, 2018-09-11:
Hi @JunhwanPark, hope you are well,

Can I just update the binary, or do I have to somehow patch slsi code too? 
We froze TizenRT version for stable development, moving to fork right now is also a lot of work - as configurations, our code etc. has to be updated. 

What worries me the most is that this fail causes permanent freeze which needs human intervention for hardware reset - this is unmaintainable in field. Is there at least any hw watchdog which we can be sure will at least reboot the platform?

JunhwanPark, 2018-09-11:
@pholat 
I am not a wifi developer, but I have seen such an option. Could you check this option?
```c
config SCSC_WLAN_AUTO_RECOVERY
	bool "Enable Auto Recovery"
	default n
	---help---
	  Enables auto recovery on FW panic and SCSC driver watch dog timeouts,
	  which will reset subsystem and re-enable state.
```

JunhwanPark, 2018-09-11:
Oh, this issue is about that option. Sorry. I misunderstood your question. 
There are a lot of changes to wifi. I can only tell you to check out the Fork version.

pholat, 2018-09-11:
It's ok - we actually checked with this option and without. 
I'll have to check how much work it will be for us to update to the fork, though we would rather use something considered stable. 

But is there any hw watchdog? We would love to have option to hard reset the board in case of terrible terrible damage, it's always  better then sending service, and we don't get any guarantee that with updated fork we won't have this issue, or even new ones.

JunhwanPark, 2018-09-11:
Yes. There is. There are also examples related to it, please check.
(apps/examples/watchdog)

pholat, 2018-09-11:
@JunhwanPark  Thanks, will try with hw watchdog first.
Then I guess we will move to last stable version of fork (right now it's: refs/tags/v1.0.12 ) - is it proper way to go? I mean I don't fully comprehend release & repository politics here.

pholat, 2018-09-11:
@JunhwanPark 
Ok :) adding wifi binary from fork didn't change much, fortunately hw watchdog works as desired - so we will stick with it for now.

JunhwanPark, 2018-09-11:
@pholat @sunghan-chang 
The Fork version https://github.com/SamsungARTIK/TizenRT release policy is as follows.

1. Major release versions will be issued in April (or May) and October each year. 
2. We then issue a monthly release of the Minor release for bug fixes and stabilization. 

The current version was released in August(18.05.03), and we are preparing for the major version in October(18.10.00).

* Major
[release/ARTIK-SW-18.05.00](https://github.com/SamsungARTIK/TizenRT/releases/tag/release%2FARTIK-SW-18.05.00) https://github.com/SamsungARTIK/TizenRT/commit/d1dfb56208b5897ac852a713b6f760ce4efdbd7a
* Minor
[release/ARTIK-SW-18.05.01](https://github.com/SamsungARTIK/TizenRT/releases/tag/release%2FARTIK-SW-18.05.01) https://github.com/SamsungARTIK/TizenRT/commit/c3fe959667afa4bf1fc8eabce17c1af34d348232
[release/ARTIK-SW-18.05.02](https://github.com/SamsungARTIK/TizenRT/releases/tag/release%2FARTIK-SW-18.05.02) https://github.com/SamsungARTIK/TizenRT/commit/325bf46a7eddfcb2657b5c0f89ef04fcb9e2bc03
[release/ARTIK-SW-18.05.03](https://github.com/SamsungARTIK/TizenRT/releases/tag/release%2FARTIK-SW-18.05.03) https://github.com/SamsungARTIK/TizenRT/commit/699bd8812ad0a6ba34db0fb02bd85f04c11bc8c2

Please refer to this schedule.
Thank you.

closed 2019-11-26 by sunghan-chang