Samsung/TizenRT #2314: Master branch source build failed.

juhaki, 2018-10-04:
juhaki@juhakidesk3:~/workspace/180927TizenRT$ git branch
* master
juhaki@juhakidesk3:~/workspace/180927TizenRT$ cat .git/config 
[core]
	repositoryformatversion = 0
	filemode = true
	logallrefupdates = true
[remote "origin"]
	url = https://github.com/Samsung/TizenRT
	fetch = +refs/heads/*:refs/remotes/origin/*
[branch "master"]
	remote = origin
	merge = refs/heads/master
juhaki@juhakidesk3:~/workspace/180927TizenRT$ git branch
* master
juhaki@juhakidesk3:~/workspace/180927TizenRT$ 
juhaki@juhakidesk3:~/workspace/180927TizenRT$ git log
commit aadb5ecb088e057f3c840f437240a19b14e30444
Merge: a5e5ffa 2dfba77
Author: sunghan-chang <sh924.chang@samsung.com>
Date:   Thu Oct 4 17:10:52 2018 +0900

    Merge pull request #2300 from manoharahk/refactor-board
    
    Refactor board directory.
...

**below info is build fail log.**
...

LD: tinyara
arm-none-eabi-ld --entry=__start -g --gc-sections -T/home/juhaki/workspace/180927TizenRT/os/../build/configs/artik053s/scripts/flash.ld -L"/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries"  \
	-o "/home/juhaki/workspace/180927TizenRT/os/../build/output/bin/tinyara" arm_vectortab.o  \
	--start-group -lkernel -lc -lmm -larch -lapps -lexternal -lnet -loctbstack -lc_common -lconnectivity_abstraction -lcoap -llogger -locsrm -lroutingmanager -lresource_directory -lfs -ldrivers -lframework chip/sss/libispdriver.a "/home/juhaki/tizen-studio/rt-ide/tools/arm-none-eabi-gcc-6.3.1/bin/../lib/gcc/arm-none-eabi/6.3.1/libgcc.a" --end-group -Map /home/juhaki/workspace/180927TizenRT/os/../build/output/bin/tinyara.map
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libarch.a(s5j_boot.o): In function `arm_boot':
/home/juhaki/workspace/180927TizenRT/os/arch/arm/src/chip/s5j_boot.c:217: undefined reference to `s5j_board_initialize'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libkernel.a(os_bringup.o): In function `os_do_appstart':
/home/juhaki/workspace/180927TizenRT/os/kernel/init/os_bringup.c:236: undefined reference to `board_initialize'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libapps.a(common_handlers.o): In function `t_things_reboot_loop':
/home/juhaki/workspace/180927TizenRT/apps/examples/st_things/common_handlers.c:382: undefined reference to `boardctl'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libapps.a(tash_command.o): In function `tash_reboot':
/home/juhaki/workspace/180927TizenRT/apps/shell/tash_command.c:188: undefined reference to `boardctl'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libexternal.a(slsi_wifi_api.o): In function `slsi_init_nvram':
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:3014: undefined reference to `up_wlan_read_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:3022: undefined reference to `up_wlan_erase_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:3035: undefined reference to `up_wlan_write_config'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libexternal.a(slsi_wifi_api.o): In function `slsi_get_country_code':
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2584: undefined reference to `up_wlan_read_config'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libexternal.a(slsi_wifi_api.o): In function `slsi_set_country_code':
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2622: undefined reference to `up_wlan_read_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2625: undefined reference to `up_wlan_erase_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2633: undefined reference to `up_wlan_write_config'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libexternal.a(slsi_wifi_api.o): In function `slsi_set_tx_power':
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2704: undefined reference to `up_wlan_read_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2707: undefined reference to `up_wlan_erase_config'
/home/juhaki/workspace/180927TizenRT/external/slsi_wifi/slsi_wifi_api.c:2711: undefined reference to `up_wlan_write_config'
/home/juhaki/workspace/180927TizenRT/os/../build/output/libraries/libapps.a(fota_hal.o): In function `fotahal_update':
/home/juhaki/workspace/180927TizenRT/apps/system/fota_hal/fota_hal.c:291: undefined reference to `boardctl'
Makefile:186: recipe for target '../build/output/bin/tinyara' failed
make[1]: *** [../build/output/bin/tinyara] Error 1
make[1]: Leaving directory '/home/juhaki/workspace/180927TizenRT/os/arch/arm/src'
Makefile.unix:464: recipe for target 'pass2' failed
make: *** [pass2] Error 2
[STATUS] FAILED

sunghan-chang, 2018-10-04:
@juhaki I think previous build results are remain with latest code. Our folder structure was changed. The board folder was splitted from arch and moved to os. But if you have build result which was made by previous structure, it can be failed.
Could you try to do `make distclean` and to remove objects manually?
```bash
find . -name *.o -delete
```
at base directory of TizenRT

sunghan-chang, 2018-10-04:
@juhaki I've tested it and it returns success and the travis (build tester) also returns success with latest codes.

juhaki, 2018-10-05:
Thanks for guide.

I have been build succeed by the below step.
$ find . -name *.o -delete
$ cd os
$ make distclean
-> Set the board type on the TizenRT IDE "Select Board" -> Set the Application Manager on the TizenRT IDE -> Set the Kernel Config on the TizenRT IDE -> "Batch Build Project"

sunghan-chang, 2018-10-05:
As you said, it works. Let's close this issue.

closed 2018-10-05 by sunghan-chang